iizukanao/picam #98: Problems using ALSA dsnoop device to share mic

dig412, 2018-02-18:
Hi - thanks very much for picam, it's a great piece of software!

I'm trying to use a `dsnoop` device in ALSA so that another application can access the microphone at the same time as picam.

Here's my `/etc/asound.conf`:
```
pcm.capture {
    type plug
    slave.pcm "array"
}
pcm.array {
    type dsnoop
    slave {
        pcm "hw:1,0"
        channels 2
    }
    ipc_key 666660
}
```
With this config I can run two copies of `arecord` at the same time without issue:
```
arecord -f cd -D capture --dump-hw-params | aplay
```

However, I can't get picam to use the same device successfully:
```
pi@babypi:~ $ /home/pi/picam  --alsadev capture
configuring devices
error: failed to set buffer size for microphone: buffer_size=2048 multiply=0 (Invalid argument)
```

Here's the relevant bits of a `--verbose` log:
```
alsa_dev=capture
audio_channels=1
audio_sample_rate=48000
audio_bitrate=40000
audio_volume_multiply=1.000000
...snip...
opening ALSA device for capture: capture
final audio_channels: 1
sample_rate: 48000
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 4
av_frame->channel_layout: 4
audio_codec_ctx->channels: 1
av_frame->channels: 0
allocated 2048 bytes for audio samples
audio_pts_step_base: 1920
actual sample rate=48000 dir=0
failed to set buffer size for microphone: buffer_size=2048 multiply=100
trying smaller buffer size for microphone: buffer_size=2048 multiply=50
failed to set buffer size for microphone: buffer_size=2048 multiply=50
trying smaller buffer size for microphone: buffer_size=2048 multiply=25
failed to set buffer size for microphone: buffer_size=2048 multiply=25
trying smaller buffer size for microphone: buffer_size=2048 multiply=12
failed to set buffer size for microphone: buffer_size=2048 multiply=12
trying smaller buffer size for microphone: buffer_size=2048 multiply=6
failed to set buffer size for microphone: buffer_size=2048 multiply=6
trying smaller buffer size for microphone: buffer_size=2048 multiply=3
failed to set buffer size for microphone: buffer_size=2048 multiply=3
trying smaller buffer size for microphone: buffer_size=2048 multiply=1
failed to set buffer size for microphone: buffer_size=2048 multiply=1
error: failed to set buffer size for microphone: buffer_size=2048 multiply=0 (Invalid argument)
```

Can you suggest anything I could do to fix this?

sinterlkaas, 2018-08-10:
The following configuration is working for me. (this is based on [http://www.alsa-project.org/main/index.php/Asoundrc#dsnoop](url))

**.asoundrc**
```
pcm.mixin {
         type dsnoop
         ipc_key 5978293 # must be unique for all dmix plugins!!!!
         ipc_key_add_uid yes
         slave {
                 pcm "hw:1,0"
                 channels 2
                 period_size 1024
                 buffer_size 4096
                 rate 44100
                 periods 0
                 period_time 0
         }
         bindings {
                 1 1
                 1 0
         }
 }
``` 

**picam start command**
/home/pi/picam/picam --samplerate 44100 --alsadev pcm.mixin

**arecord command**
 `arecord -d 30 -f cd -t wav -D pcm.mixin test.wav`

sergioaiello, 2019-03-25:
Hi, I hope it's not late to jump into this thread.
And I hope my question is appropriate here.
I'm recording audio from the microphone of an
Linux orangepizeroplus15 4.19.20-sunxi64 #5.75 SMP Fri Feb 8 10:29:25 CET 2019 aarch64 GNU/Linux
and I cannot stream audio on demand. My need is related to a monitoring solution.
Creating a dsnoop interface could be the solution, but do not how. Even because I've found 
/usr/share/alsa/alsa.conf but not /etc/asound.conf.
Thank you!