iizukanao/picam #100: HLS Stream with Clappr-Player

lippih, 2018-03-10:
I would like to use a simple player from Clappr to view the HLS-stream from picam. The Clappr Player works well for all HLS Stream I found on the web, but it doesnt work  for the picam Stream. So my questions are: 

1) Is the the picam Stream (HLS) not "standard"?
2) How can I change the specification for the HLS Stream from picam?

Thank you

iizukanao, 2018-03-11:
picam uses standard HLS. See `picam --help` for available options to configure HLS.

Do you see errors like this in your browser's console?
```
Failed to load http://pi/index.m3u8: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://...' is therefore not allowed access.
log.js:65 [error][hlsjs: unrecoverable network fatal error, evt hlsError, data [object Object] ]
```

If so, change the configuration of your web server so that response header of index.m3u8 contains `Access-Control-Allow-Origin: *`. The response should look like this:
```
% curl -i http://pi/index.m3u8
HTTP/1.1 200 OK
Date: Sun, 11 Mar 2018 02:45:17 UTC
Connection: close
Content-Length: 145
Content-Type: application/x-mpegURL
Access-Control-Allow-Origin: *

#EXTM3U
#EXT-X-VERSION:3
#EXT-X-TARGETDURATION:1
#EXT-X-MEDIA-SEQUENCE:492
#EXT-X-ALLOW-CACHE:NO
#EXTINF:1.00000,
491.ts
#EXTINF:1.00000,
492.ts
```

lippih, 2018-03-11:
Thank you for the reply!
I did insert follow srtings in /etc/nginx/sites-available/default:

```
location / {
  add_header 'Access-Control-Allow-Origin' '*';
  add_header 'Access-Control-Allow-Credentials' 'true';
  add_header 'Access-Control-Allow-Methods' 'GET, POST, OPTIONS';
  root /run/shm;
}
```

now i can look the stream! But if I look my index.m3u8 it looks like before:
```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-TARGETDURATION:1
#EXT-X-MEDIA-SEQUENCE:3107
#EXT-X-ALLOW-CACHE:NO
#EXTINF:1.00000,
3106.ts
#EXTINF:0.97333,
3107.ts
```
Why I have not the header like in your example?

Just for interest, the code of my simple player is this:

```
<!DOCTYPE html>
<head>
  <script type="text/javascript" src="http://192.168.2.53:8083/clappr/dist/clappr.min.js"></script>
</head>
<body>
  <div id="player"></div>
  <script>
    var player = new Clappr.Player({source: "http://192.168.2.53:8083/hls/index.m3u8", parentId: "#player",
    baseUrl: "http://192.168.2.53:8083/clappr"    
    });
  </script> 
</body>

```

lippih, 2018-03-12:
Yes it is solved, thank you again!

closed 2018-03-12 by lippih