iizukanao/picam #101: Running picam as a bash script

sakshirawal14, 2018-03-11:
How can I run the following commands as a bash script. When I call the following commands from a bash script: 
$ ./picam --alsadev hw:1,0 &
touch hooks/start_record
touch hooks/stop_record
It executes the first command, prints the following statements and stalls there. The execution does not proceed to the start record command.
configuring devices
capturing started

Could anyone please let me know what am I missing?

iizukanao, 2018-03-11:
You need to wait for `capturing started` message in order to start recording.

iizukanao, 2018-03-11:
You need to wait for `capturing started` message in order to start recording.

The script will look like this:
```
#!/bin/sh
# Record 10 seconds of video

echo ">> starting picam"
./picam --alsadev hw:1,0 &
sleep 3

echo ">> begin recording"
touch hooks/start_record
sleep 10

echo ">> finish recording"
touch hooks/stop_record
sleep 1
killall picam

echo ">> done"
```

sakshirawal14, 2018-03-11:
Thanks a ton @iizukanao for your timely revert. I'll try the suggested script and keep you posted on the results.

Highly grateful :)

sakshirawal14, 2018-03-12:
Hi @iizukanao

The code works perfectly fine :)
Except the "kill picam" command on the second-last line. It is unable to kill picam and gives the error:
./script.sh: 13: kill: Illegal number: picam

It is expecting the process ID for picam. How can this be resolved?

Can't thank you enough for all the help rendered

iizukanao, 2018-03-12:
It is `killall` not `kill`. `killall` expects a process name but `kill` expects a PID.
In case of you need picam's PID, use `pidof picam`.

sakshirawal14, 2018-03-12:
Thanks @iizukanao :)
Works perfectly fine. 

Have been testing the code since morning and the only error I face now at the end of the day and last few of my MPEG recordings is the microphone buffer overrun.

Any points on how to resolve this and why does this error occur.

sakshirawal14, 2018-04-05:
Hi @iizukanao
Hope you are doing well. Thanks a ton for helping me out with the implementation of your library. Can't thank you enough.

However, I need a small help from you yet again. For recording a video for 10 seconds, I can't use the sleep function (reason being it takes the terminal into a blocking state). Could you please suggest how can I record a 10 second video and then stop recording (without using sleep). May be I can run this section with a timeout. Have tried a few ways but no luck yet.

iizukanao, 2018-05-07:
If your Python script executes a shell script that contains `sleep`, use [subprocess.Popen()](https://docs.python.org/3/library/subprocess.html#subprocess.Popen) in order to avoid blocking.

closed 2018-05-07 by iizukanao

feacluster, 2019-02-28:
Many thanks! I spent several hours trying to figure out why my bash script would not work. As you mentioned I need a sleep command after executing picam and before the touch hooks/start_record ! Perhaps this should be mentioned in the original documentation?