iizukanao/picam #103: Increasing dts to muxer error

tomph, 2018-03-24:
Hi there,  I've been experiencing some stream freeze issues (on the client side at least), using both a web player and an app I've built (Android/Exoplayer).

The picam log is reporting the following:

"[mpegts @ 0x2b46560] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 17705050 >= 1"

...Which could be related to (https://stackoverflow.com/questions/23408309/avconv-complains-non-monotonically-increasing-dts-to-muxer-in-stream-when-pipi)

I'm using the following settings with picam:

"./picam --alsadev hw:1,0 -c 1 --volume 0.1 -r 44100 -a 64000 --vfr --maxfps 30 --minfps 10 --shutter 20000 -v 1000000 -w 1024 -h 768 --hflip --vflip "

Any ideas what this might be?

ADoncel, 2019-05-16:
I'm having the same issue, did you manage to discover a solution or at least the reason?

Thank you.

tomph, 2019-05-16:
Sadly no.  I managed to reduce freezes by tweaking lots of settings in my app, but never managed to completely solve it :(