iizukanao/picam #104: error: failed to set camera to idle state

ErHaMep, 2018-04-04:
Hi, 
It seems you have a very nice program who combines video and audio.

But in my RPI it does not work.
I have installed your SD-card image and run picam.
Then it shows this error:
configuring devices
error: failed to set camera to idle state
Perhaps another program is using camera, otherwise you need to reboot this pi

But after a reboot there is the same error.
On internet I found no examples of this error, only in stream.c of picam.

I use this camera because it has adjustable focus.
[https://www.waveshare.com/rpi-camera-b.htm
](url)

I have not tried it with the standard RPI camera.

Is there something more to do?

Best regards,
Roelof

paolohu, 2018-04-04:
So - I don't know if this is related or not - but I have had it happen so I'll throw it out there...

What are you using to power the RPI?  I had an issue before on one RPI that I would continually get that error - checked the camera over and over - no issues.  Turns out the power supply I was using was giving the board enough power to turn on but could not power the camera.  I swapped out the power supply to something that gave the proper volts/amps and it worked fine.

The other standard stuff I'd check is that the camera is indeed enabled in raspiconfig and that all the pins on that ribbon cable are facing the correct way and are making proper contact.  If you swapped out the ribbon cable make sure it's the correct one (the RPI cable has the leads on opposite sides on each side of the ribbon cable - it's easy to grab one off amazon and get the wrong type).

ErHaMep, 2018-04-06:
Hello,

The power supply I use is a standard RPI power supply for the RPi 3, 5 V 2.5 A and with a RPi sign on it.

Before I tried picam I was busy with raspivid and cvlc.
`#raspivid -md 4 -w 1296 -h 972 -p 960,50,320,240 -t 0 -o - | cvlc -vvv stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/}' :demux=h264`
This works good, but I want to include audio. 

I think there may be a problem with the design and made in China camera. This camera works with a OV5647 sensor.
So I ordered a standard RPi camera and will try that with picam. This is a V2 model with IMX219 sensor.

I have copied the image on the SD card, put it in the rpi and run picam. It that enough to run picam?
For the working of the camera is there a need to run make_dirs.sh also? This file has more to done with the streaming and so on?

Thanks for your help.
Roelof

ErHaMep, 2018-09-17:
Hello,

I have checked the standard RPi 3 powersupply with an USB voltage and current monitor, the UM25C.
There are no problems with the power for RPi 3.
Also the waveshare.com camera and standard V2 camera works in the same way.
It seems there is no hardware problem in my system. See also my remarks earlier.

But today I see that I put this issue in the wrong repository.
I am using picam-streamer so I will stop here and make a new issue in the picam-streamer repository.

Thanks so far,
Roelof

synaption, 2020-08-15:
This is a pretty old thread, but I will add that I solved the problem by running
$ sudo service motion start
$ sudo service motion stop
I am sure that motion was not running or interfering.  I have to assume that motion does something to reset everything clearly.

feacluster, 2021-03-31:
I have seven of these cameras and get this error with one camera every few days. It is random though . Sometimes I need to reboot every day. Other times it goes fine for a few weeks. I am recording video upto 8 hours a day with them. Perhaps some hardware or power bug?