iizukanao/picam #109: Is it possible to use different cameras that aren't USB?

DanPugly, 2018-06-22:
I was going to ask if I can use USB cameras but then I saw another post before, but my question is can I use different cameras that aren't USB such as this one maybe?: https://www.amazon.com/Arducam-Megapixels-OV5647-Camera-Raspberry/dp/B012ETE75I/ref=sr_1_7?ie=UTF8&qid=1529670784&sr=8-7&keywords=pi+camera

umolac, 2018-06-22:
Well, since it uses CSI, and has specification really similar to Camera V1, I don't think that shouldn't be possible

DanPugly, 2018-06-22:
yeah but like in principle, I took the first camera that was similar to the raspberry pi cam on Amazon :D
so it shouldn't have any problems...

iizukanao, 2018-06-23:
Sorry, but USB cameras are not supported.

DanPugly, 2018-06-23:
I didnt mean a USB cam i meant a cemra that uses the same connection as the
official raspberry pi cam?

iizukanao, 2018-06-24:
@DanPugly Oh, sorry for misunderstanding. I think that Arducam camera works because it uses the same interface (CSI connector and OV5647 image sensor) as Raspberry Pi Camera Module v1.

DanPugly, 2018-06-24:
Ok thanks :D

DanPugly, 2018-08-11:
Is the official camera v2 supported?

On Jun 24, 2018 10:00, "Pugly" <dushomirsky@gmail.com> wrote:

Ok thanks :D

iizukanao, 2018-08-13:
@DanPugly Yes, Camera Module V2 is supported.