iizukanao/picam #111: Does no longer work after latest apt-get upgrade?

Zaldoran, 2018-08-30:
I've had a working installation of picam recently, and figured out that it does no longer work after latest upgrade. After starting picam, it repeats

`warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.`

while changing parameters does not have any effect. When connecting to stream, only audio is streamed, but no video anymore. I did reinstall everything twice with the same results. Finally, I'd installed the picam streamer sd-card image which did work initially, but same issue after 

`sudo apt-get upgrade`

(however, it does not show the console warning message as mentioned above, but video isn't streamed anymore)

joenathan7, 2018-09-05:
I'm seeing the same thing with my new Pi 3+

zr0t, 2018-09-05:
Yup, same here as well.

iizukanao, 2018-09-06:
I've modified the code for the latest firmware in [fix-novideo](https://github.com/iizukanao/picam/tree/fix-novideo) branch. Can someone compile it and check if it works?

joenathan7, 2018-09-06:
I can give it a try this evening.

joenathan7, 2018-09-07:
Didn't realize this needed a separate linux machine to compile ffmpeg.  I don't have one easily set up for that, just the Pi.  There any way I can get a tag or release that's compiled to try?

iizukanao, 2018-09-07:
@joenathan7 You don't need another machine if you have Raspberry Pi 2 or above. You can build them all on your Pi. To do so, please follow these steps:

1. `sudo apt-get install libfontconfig1-dev libharfbuzz-dev`
2. Download, build and install fdk-aac
3. Download, build and install ffmpeg with `--enable-libfdk-aac` option (i.e. `./configure --enable-libfdk-aac`)
4. Build libilclient
5. Build picam

Please refer to [INSTALL.md](https://github.com/iizukanao/picam/blob/master/INSTALL.md) for detailed steps.

iwanttobefreak, 2018-09-11:
Hello iizukanao, thank you for your help!
I have raspberry pi3+
apt-get upgrade
I follow your steps but when I download fix-novideo branch and try to compile I get this error:
stream.c:32:34: fatal error: libavformat/avformat.h: not found

Thanks!

joenathan7, 2018-09-11:
Couple things
The instructions for crosstool-ng.  When you follow them and download 1.22.0 it references an address during install that apparently is lacking the correct security cert and is throwing an error when it's trying to download dependencies.  I installed 1.23 and that worked fine, however a lot of the setting versions that are suggested in the Install.md do not exist in 1.23 an when I get to the ct-ng build step it throws an error that suggests there's a bad ARM version, maybe it's not built yet for the raspPi 3rd gen?

If I just move past trying to compile crosstool-ng, I can build fdk-aac and ffmpeg and get all the way to the point where I run ffmpeg with the codecs argument and it throws an error saying 

./ffmpg: error while loading shared libraries: libfdk-aac.so.1: cannot open shared object file: No such file or directory.  

It looks like I compiled it and that library file exists in ~/pi/build/lib.  I've tried copying that file over to the bin directory so ffmpeg finds it and still no dice.

wozz, 2018-09-11:
@joenathan7 I was able to use crosstool-ng 1.22.0 by substituting:
```
https://releases.linaro.org/${YYMM}/components/toolchain/gcc-linaro/${linaro_series} \
```
for
```
https://releases.linaro.org/archive/${YYMM}/components/toolchain/gcc-linaro/${linaro_series} \
```
in `/opt/cross/lib/crosstool-ng-1.22.0/scripts/build/cc/100-gcc.sh`

However after following the instructions, and running picam, I get the same error after running for a few minutes.

joenathan7, 2018-09-12:
That fix got me past downloading the dependencies of Crosstool.  I'm still getting and error in the first pass of the gcc compiler 

make[2]: *** [s-automata] Error 137

wozz, 2018-09-16:
I created an ansible playbook that uses an AWS spot instance for the cross compile component: https://gist.github.com/wozz/2030f8a326e6d6a7801b7a98c5136a3c

@iizukanao is there anything else I can try? it doesn't seem like this fix works.

iizukanao, 2018-09-16:
I've fixed this issue and released [version 1.4.7](https://github.com/iizukanao/picam/releases/tag/v1.4.7). Please give it a try.
Thank you all for reporting the issue!

wozz, 2018-09-16:
works for me! thanks.

iizukanao, 2018-09-16:
@iwanttobefreak @joenathan7 @wozz Thanks for trying to build picam! I've updated [build instructions](https://github.com/iizukanao/picam/blob/master/BUILDING.md) to make cross-compiling unnecessary for Raspberry Pi 2 or 3 users.

joenathan7, 2018-09-17:
This worked great! Thanks guys!

matfior, 2018-10-11:
Hi there,
I am getting the same error. I have version 1.4.7, 
I have tried playing around with gopsize but it does not solve the issue.

iizukanao, 2019-07-16:
Fixed another bug in [v1.4.8](https://github.com/iizukanao/picam/releases/tag/v1.4.8). Please open a new issue if you are still having problem.

closed 2019-07-16 by iizukanao