iizukanao/picam #112: Picam preview is over all windows

iwanttobefreak, 2018-09-07:
Hi. Thank you for the project, is amazing.
picam preview is over all windows.
It's possible to run in a background layer?
I want to run
picam -p
and play a mplayer movie above the preview window. Always is bellow
Thanks

iwanttobefreak, 2018-09-08:
Thanks.
Is possible to play bellow other window?
Thanks

iizukanao, 2018-09-09:
@iwanttobefreak That's impossible.

nalajcie, 2018-09-09:
Actually, playing video above the picam may be possible, but You have to
check it by Yourself as I can't test it right now.

Everything depends on Pi HW accelerated API (OpenMAX). This API supports
multiple surfaces on different layers. The picam uses layers 14, 15 and 31
for different parts of preview window (background, video and text to be
rendered). These values are defined in this file:
https://github.com/iizukanao/picam/blob/master/dispmanx.h

For video playing You can use "omxplayer" which is mplayer modified
specifically for Pi:
https://elinux.org/Omxplayer

It is possible to change the video playback overlay level by "--layer"
commandline parameter.

Try setting the layer to 32.

iizukanao, 2018-09-10:
@nalajcie Good catch! Thanks for the info.