iizukanao/picam #113: 4 channel microphone

iizukanao, 2018-09-22:
picam should support 4 channel microphone.
requested by @leomen3
Previous discussion: https://github.com/iizukanao/picam-streamer/issues/9

iizukanao, 2018-09-22:
@leomen3 I have created a test version which accepts `-c 4`. Can you give it a try?
https://github.com/iizukanao/picam/releases/tag/v1.4.7-dev-4ch

TomUser1, 2018-09-24:
Thanks For Your Attention iizukanao,
Results:
$./picam --alsadev hw:1,0 -c 4
configuring devices
error: cannot set sample format (Invalid argument).

4 channel microphone (http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/) by default works on S32_LE simple format, but picam support only S16_LE. Also picam have no parameter for change audio simple format.
Сould you add support of this format? 
Or maybe do you know how to change default format from S32_LE to S16_LE for mic.

Thanks.

Dave4675, 2018-11-02:
> Thanks For Your Attention iizukanao,
> Results:
> $./picam --alsadev hw:1,0 -c 4
> configuring devices
> error: cannot set sample format (Invalid argument).
>
> 4 channel microphone [...]

I have adafruit microphones too, but I'm using them stereo, now the thing is that the standard adafruit asoundrc configuration is not compatible with streaming because it uses S32_LE format by default, so you need to use alsa plug-in in order to have a linear conversion, in this way the alsa module will adapt automatically to the need of the stream. Anyway, this is my asoundrc configuration file

NOTE: Take care of two things, first, the mono configuration will result into audio non synched, two, this issue doesn't mean anything with picam sofware itself:

pcm_slave.sl2 {
pcm "plughw:1,0"
format S32_LE
rate 48000
}

pcm.!default {
type rate
slave sl2
}

ctl:!default {
type rate
slave sl2
}

Now your device will be "default", also think about that the virtual soundcard module for the i2s microphones create an overhead, so this is the reason because I'm not using something that can me let adjust the volume, I suggest to you to use picam command instead.

rajeshkrarora1, 2018-11-02:
after configuring .asoundrc,   before you can change the volume you need to
use the device once (this is an alsa thing), so  run "arecord -D dmic_sv
-c2 -r 44100 -f S32_LE -t wav -V mono -v file.wav", And cancel with ^C
once  it starts recording
Now you can run alsamixer - press F6 and select the I2S simple sound card ,

It will complain there are no playback controls (because its for recording
only).
Press F5 to change the volume.

rajeshkrarora1, 2018-11-05:
> I will buy ReSpeaker 4-Mic Array and look into this issue.

sorry i am on wrong thread, because, the .asoundrc configuration and about
using soft volume control i have mentioned is in case of *adafruit* i2s
mems mic *SPH0645* and *not* in case of *SAEEED'S *4-mic array

maeffew, 2019-03-12:
Has anyone had any further success with using the respecter mic here?

mrceresa, 2020-03-26:
Hi: I'm also interested in knowing the current status of this. Thanks!

inseltraeumer, 2020-05-25:
Picam wrks with the 4 channel microphone mentioned above (the one by seeed) for me with the following commands after following the setup guide on https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/. Note that you''ll have to define the $parameters and the 

`FFREPORT=file="ffreport-$(date).log":level=32 /home/pi/ffmpeg/ffmpeg \
  -re \
  -i tcp:127.0.0.1:8181?listen \
  -filter:a aresample=async=1000 \
  -c:a aac \
  -c:v copy \
  -f flv rtmp://a.rtmp.youtube.com/live2/$1 &

./picam \
  --alsadev plughw:1,0 \
   -c $audio_channels \
   -r $audio_framerate \
   -a $audio_bitrate \
  --time \
  --timeformat "$timeformat" \
  --hflip \
  --vflip \
   -w $vwidth \
   -h $vheight \
  --ex $exposure \
  --tcpout tcp://127.0.0.1:8181

touch hooks/start_record
`

harryjubb, 2020-06-27:
@inseltraeumer Thank you for your reply, also using the ReSpeaker 4 Mic Array and have had success setting `-c 1`.

The key part for us was also setting `--alsadev plughw:1,0` rather than `--alsadev hw:1,0`.

Have not yet tried a long stream so haven't seen any buffer overrun issues, but will reply again if we can reproduce this.

Running with:

```./picam --alsadev $ALSA_DEVICE -c $AUDIO_CHANNELS --tcpout tcp://ffmpeg:8181```

Where

```ALSA_DEVICE=plughw:1,0
AUDIO_CHANNELS=1```

With ffmpeg running in a Dockerised service.

inseltraeumer, 2021-02-27:
> @inseltraeumer Thank you for your reply, also using the ReSpeaker 4 Mic Array and have had success setting `-c 1`.
> 
> 
> 
> The key part for us was also setting `--alsadev plughw:1,0` rather than `--alsadev hw:1,0`.
> 
> 
> 
> ~Have not yet tried a long stream so haven't seen any buffer overrun issues, but will reply again if we can reproduce this.~
> 
> 
> 
> *Have managed to reproduce the buffer overrun quite quickly!* Presumably related to #34? Any thoughts or advice on this would be appreciated.
> 
> 
> 
> Running with:
> 
> 
> 
> ```./picam --alsadev $ALSA_DEVICE -c $AUDIO_CHANNELS --tcpout tcp://ffmpeg:8181```
> 
> 
> 
> Where
> 
> 
> 
> ```
> 
> ALSA_DEVICE=plughw:1,0
> 
> AUDIO_CHANNELS=1
> 
> ```
> 
> 
> 
> With ffmpeg running in a Dockerised service.

I managed to circumvent this by doing 

while true; do
  if ! pgrep ffmpeg > /dev/null ; then
    # kill picam process
    sleep 13
    # start picam again
  fi
  sleep 2
done

hanskenb, 2021-03-08:
Dear All, 

i'm getting the following: "cannot open audio capture device 'plughw:1,0': No such file or directory".

arecord -l results in 

```

card 2: seeed4micvoicec [seeed-4mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0 [bcm2835-i2s-ac10x-codec0 ac10x-codec.1-003b-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
Any idea?