iizukanao/picam #114: 4 channel mic

TomUser1, 2018-09-24:
Thanks For Your Attention iizukanao,
Results:
$./picam --alsadev hw:1,0 -c 4
configuring devices
error: cannot set sample format (Invalid argument).

4 channel microphone (http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/) by default works on S32_LE simple format, but picam support only S16_LE. Also picam have no parameter for change audio simple format.
Сould you add support of this format?
Or maybe do you know how to change default format from S32_LE to S16_LE for mic.

Thanks.

Dave4675, 2018-11-02:
I have adafruit microphones too, but I'm using them stereo, now the thing is that the standard adafruit asoundrc configuration is not compatible with streaming because it uses S32_LE format by default, so you need to use alsa plug-in in order to have a linear conversion, in this way the alsa module will adapt automatically to the need of the stream. Anyway, this is my asoundrc configuration file

NOTE: Take care of two things, first, the mono configuration will result into audio non synched, two, this issue doesn't mean anything with picam sofware itself:

pcm_slave.sl2 {
pcm "plughw:1,0"
format S32_LE
rate 48000
}

pcm.!default {
type rate
slave sl2
}

ctl:!default {
type rate
slave sl2
}

Now your device will be "default", also think about that the virtual soundcard module for the i2s microphones create an overhead, so this is the reason because I'm not using something that can me let adjust the volume, I suggest to you to use picam command instead.

Dave4675, 2018-11-02:
This is only a hint in your specific case