iizukanao/picam #116: issues in ct-ng build, in cross compilation for raspberry pi zero

rajeshkrarora1, 2018-11-14:


rajeshkrarora1, 2018-11-14:
my build.log looks like this

[INFO ]  Performing some trivial sanity checks
[DEBUG]  Testing '! ( -n  )'
[DEBUG]  Testing '! ( -n  )'
[DEBUG]  Testing '! ( -n  )'
[DEBUG]  Testing '! ( -n  )'
[DEBUG]  Testing '! ( -n  )'
[DEBUG]  Testing '! ( -n  )'
[DEBUG]  ==> Executing: 'mkdir' '-p' '/home/pi/pi/ctng/.build' 
[DEBUG]  ==> Executing: 'rm' '-f' '/home/pi/pi/ctng/.build/backtrace' 
[DEBUG]  ==> Executing: 'touch' '/home/pi/pi/ctng/.build/foo' 
[DEBUG]  Testing '! ( -f /home/pi/pi/ctng/.build/FOO )'
[DEBUG]  ==> Executing: 'rm' '-f' '/home/pi/pi/ctng/.build/foo' 
[DEBUG]  Testing '-f /bin/bash -a -x /bin/bash'
[DEBUG]  Creating bin-override for tools in '/home/pi/pi/ctng/.build/tools'
[DEBUG]  ==> Executing: 'mkdir' '-p' '/home/pi/pi/ctng/.build/tools/bin' 
[DEBUG]  Creating script-override for 'install' -> '"/usr/bin/install -c"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/install' 
[DEBUG]  Creating script-override for 'bash' -> '"/bin/bash"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/bash' 
[DEBUG]  Creating script-override for 'awk' -> '"/usr/bin/gawk"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/awk' 
[DEBUG]  Creating script-override for 'grep' -> '"/bin/grep"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/grep' 
[DEBUG]  Creating script-override for 'make' -> '"/usr/bin/make"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/make' 
[DEBUG]  Creating script-override for 'sed' -> '"/bin/sed"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/sed' 
[DEBUG]  Creating script-override for 'libtool' -> '"/usr/bin/libtool"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/libtool' 
[DEBUG]  Creating script-override for 'libtoolize' -> '"/usr/bin/libtoolize"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/libtoolize' 
[DEBUG]  Creating script-override for 'objcopy' -> '"/usr/bin/objcopy"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/objcopy' 
[DEBUG]  Creating script-override for 'objdump' -> '"/usr/bin/objdump"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/objdump' 
[DEBUG]  Creating script-override for 'readelf' -> '"/usr/bin/readelf"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/readelf' 
[DEBUG]  Creating script-override for 'patch' -> '"/usr/bin/patch"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/patch' 
[DEBUG]  Creating script-override for 'gperf' -> '"/usr/bin/gperf"'
[DEBUG]  ==> Executing: 'chmod' '700' '/home/pi/pi/ctng/.build/tools/bin/gperf' 
[INFO ]  Build started 20181114.064208
[DEBUG]  =================================================================
[DEBUG]  Dumping user-supplied crosstool-NG configuration
[DEBUG]    ==> Executing: '/bin/grep' '-E' '^(# )?CT_' '.config' 
[DEBUG]    CT_CONFIGURE_has_xz=y
[DEBUG]    CT_MODULES=y
[DEBUG]    # CT_OBSOLETE is not set
[DEBUG]    CT_EXPERIMENTAL=y
[DEBUG]    # CT_ALLOW_BUILD_AS_ROOT is not set
[DEBUG]    # CT_DEBUG_CT is not set
[DEBUG]    CT_LOCAL_TARBALLS_DIR="/home/pi/pi"
[DEBUG]    CT_SAVE_TARBALLS=y
[DEBUG]    CT_CUSTOM_LOCATION_ROOT_DIR=""
[DEBUG]    CT_WORK_DIR="${CT_TOP_DIR}/.build"
[DEBUG]    CT_PREFIX_DIR="$/opt/cross/x-tools/${CT_TARGET}"
[DEBUG]    CT_INSTALL_DIR="${CT_PREFIX_DIR}"
[DEBUG]    CT_RM_RF_PREFIX_DIR=y
[DEBUG]    CT_REMOVE_DOCS=y
[DEBUG]    CT_INSTALL_DIR_RO=y
[DEBUG]    CT_STRIP_HOST_TOOLCHAIN_EXECUTABLES=y
[DEBUG]    # CT_STRIP_TARGET_TOOLCHAIN_EXECUTABLES is not set
[DEBUG]    # CT_FORBID_DOWNLOAD is not set
[DEBUG]    # CT_FORCE_DOWNLOAD is not set
[DEBUG]    CT_CONNECT_TIMEOUT=10
[DEBUG]    # CT_ONLY_DOWNLOAD is not set
[DEBUG]    # CT_USE_MIRROR is not set
[DEBUG]    # CT_FORCE_EXTRACT is not set
[... 1196 lines cut ...]
[DEBUG]    Trying 'http://cbuild.validation.linaro.org/snapshots/gcc-linaro-4.9-2015.06'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/pi/pi/ctng/.build/tarballs/gcc-linaro-4.9-2015.06.tmp-dl' 'http://cbuild.validation.linaro.org/snapshots/gcc-linaro-4.9-2015.06' 
[ALL  ]    --2018-11-14 06:42:25--  http://cbuild.validation.linaro.org/snapshots/gcc-linaro-4.9-2015.06
[ALL  ]    Resolving cbuild.validation.linaro.org (cbuild.validation.linaro.org)... 51.148.40.7
[ALL  ]    Connecting to cbuild.validation.linaro.org (cbuild.validation.linaro.org)|51.148.40.7|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... 302 Found
[ALL  ]    Location: https://cbuild.validation.linaro.org/snapshots/gcc-linaro-4.9-2015.06 [following]
[ALL  ]    --2018-11-14 06:42:25--  https://cbuild.validation.linaro.org/snapshots/gcc-linaro-4.9-2015.06
[ALL  ]    Connecting to cbuild.validation.linaro.org (cbuild.validation.linaro.org)|51.148.40.7|:443... connected.
[ALL  ]    The certificate's owner does not match hostname 'cbuild.validation.linaro.org'
[DEBUG]    Not at this location: "http://cbuild.validation.linaro.org/snapshots/gcc-linaro-4.9-2015.06"
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_gcc_get[scripts/build/cc/100-gcc.sh@30]
[ERROR]  >>        called from: do_cc_get[scripts/build/cc.sh@21]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@593]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/opt/cross/share/doc/crosstool-ng/crosstool-ng-1.22.0/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 0:17.76)

rajeshkrarora1, 2018-11-14:
HI Dear iizukanao, thank you for this wonderful project, picam-streamer works wonderfully fine with raspberry pi 3, but with raspberry pi zero it works without audio, i'am using usb mic, as i2s mics are not working with "picam-streamer"   , so i'am trying to build it using "cross- compilation method described by yourself, i'am have trouble in ct ng build step, the build log has all the description, the major error i see says "Build failed in step "[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_gcc_get[scripts/build/cc/100-gcc.sh@30]
[ERROR]  >>        called from: do_cc_get[scripts/build/cc.sh@21]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@593]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/opt/cross/share/doc/crosstool-ng/crosstool-ng-1.22.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 0:17.76)
[00:18] / /opt/cross/bin/ct-ng:152: recipe for target 'build' failed
make: *** [build] Error 1

**pls help me with this**

rajeshkrarora1, 2018-11-14:
i am using **Raspbian stretch** installed on my **PC**, which has **I3 intel processor, DH55TC intel mother board and 4GB Ram, 1TB sata storage**

rajeshkrarora1, 2018-11-15:
sir,
 i have not understood , when to start working on raspberry pi zero for the first time during the whole process, do i need to "ct-ng build on raspberry pi zw or the non-rpi  linux machine,  **pls put some light**

iizukanao, 2018-11-16:
@rajeshkrarora1 cross-compiling with ct-ng is too complicated and I recommend you follow [BUILDING.md](https://github.com/iizukanao/picam/blob/master/BUILDING.md) instead. It takes long time to finish but the whole process is much simpler.

rajeshkrarora1, 2018-11-16:
Sir, i need to use raspi zero w only

iizukanao, 2018-11-16:
@rajeshkrarora1 You can build everything on Pi Zero W only.

rajeshkrarora1, 2018-11-16:
Sir, do you mean by using BUILDING.Md for pi z?

rajeshkrarora1, 2018-11-16:
Sir, i also need to use i2s mic and not a usb one.

iizukanao, 2018-11-16:
@rajeshkrarora1 Yes, you can use BUILDING.md for Pi Zero W. I2S mic is another issue, but if it's supported by ALSA, I think you can use it with picam.

rajeshkrarora1, 2018-11-16:
Sir, mic is supported by ALSA,and also i'm able to change its volume in
ALSA using softvol control, but sound is not heard in the stream, though
when i stream only audio using cvlc using the same .asoundrc file it works
fine

rajeshkrarora1, 2018-11-16:
sir, for audio streaming with i2s mems mic  i'am not using picam but after
configuring i2s mic on raspbian image( as per
https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-and-test)
and than making some changes in .asoundrc file. I install vlc on it and
then use cvlc command to stream audio which works fine on rpi zero as well
as rpi3

 In "picam-streamer", usb mic works but i2s mems mic does not. (only with
rpi3 not rpi zero).

for rpi3 i following your instructions using Building.md, before installing
"ffmpeg" i followed instructions at "https://github.com/nejohnson2/rpi-i2s",
for adding I2S mic to rpi, but still not sure how to configure ffmpeg in a
way that picam take audio input from I2S mic
*Pls Help*

rajeshkrarora1, 2018-11-16:
Sorry, closed this issue by mistake

rajeshkrarora1, 2018-11-25:
sir, with raspberry pi zero, in PICAM-STREAMER ,audio does not work, whether it is usb, or i2s mic