iizukanao/picam #118: warning: Record buffer is starving - version 1.4.7-4ch

frankmundt, 2018-11-25:
Im running picam 1.4.7-4ch on Raspberry Pi B+.  I've read a few postings and have tried different combinations of settings with no luck.  Currently running picam with default settings "picam -o /run/shm/hls  --autoex --time --alsadev hw:1,0  --verbose" and I've attached a log file as well.  The only other process that is running on the Pi is nginx as a webserver.  I have added crontab to restart the process as soon as it sees the error in the log file. 

[picam.log](https://github.com/iizukanao/picam/files/2613522/picam.log)

mprecup, 2018-12-13:
I'm running 1.4.6 on a Raspberry Pi 3 B+ and get the same error with the same command.

iizukanao, 2019-07-16:
This bug has been fixed in [v1.4.8](https://github.com/iizukanao/picam/releases/tag/v1.4.8). Sorry for taking a long time to fix.

closed 2019-07-16 by iizukanao

mprecup, 2019-07-16:
No worries! Thank you!