iizukanao/picam #120: How to take a picture while recording video?

AmirIR, 2019-01-15:
Hi,
How can I take pictures for while recording video?

AmirIR, 2019-01-17:
Ok, Thanks :(

closed 2019-01-17 by AmirIR

feacluster, 2021-01-27:
There is a way to do it by using ffmpeg to extract the last frame from the temporary video files that picam creates. It is not 100% reliable, but close enough. I can share the code if interested.

AmirHosseinCV, 2021-01-28:
> There is a way to do it by using ffmpeg to extract the last frame from the temporary video files that picam creates. It is not 100% reliable, but close enough. I can share the code if interested.

I used ffmpeg + picamera instead of this project. Thanks for the reply!

toshibochan, 2021-01-28:
Hi @feacluster please you can share your code. Thank you

feacluster, 2021-01-28:
> Hi @feacluster please you can share your code. Thank you

Try this php code:

$out = shell_exec ( "cat logs/run_picam_output.txt | egrep '\.ts$'" );

if ( preg_match ( '/(rec\/.*?\.ts$)/', $out , $matches ) ) $file = $matches[1];
    else { "Echo no match found for $out\n"; return ; }

echo "Temporary video file is at: $file\n";
$out_file = "/var/www/html/share/$hostname.jpg";
$temp_out_file = $out_file . '_temp.jpg';

if ( filesize ( $out_file ) > 0 ) shell_exec ( "cp $out_file $temp_out_file" );

$cmd = "ffmpeg -y -sseof -3 -i $file -vsync 2 -update 1 -vf scale=640:480 -q:v 1 /tmp/test.jpg ";
$pid = exec ( $cmd . ' > /dev/null 2>&1 & echo $!; ' , $output);
echo "Process ID is $pid\n";
sleep(3);
exec ( "kill -9 $pid");
//sleep(3);

$size = filesize ( '/tmp/test.jpg' );
$temp_size = filesize ( $temp_out_file );
echo "Filesize is $size, Temp filesize is $temp_size";

if ( intval ( $size ) < 1 ) shell_exec ( "cp $temp_out_file $out_file" );
    else exec ( "cp /tmp/test.jpg $out_file" );

unlink ( $temp_out_file );