iizukanao/picam #121: picam with nginx-rtmp-module issue

viveksanagari, 2019-01-25:
Hi,
I am trying to use picam with the nginx-rtmp-module (i.e, [https://github.com/iizukanao/picam#using-picam-in-combination-with-nginx-rtmp-module](url) ).
I want to make use of nginx-rtmp-module for its Nginx HTTP control module features (i.e, [https://github.com/arut/nginx-rtmp-module/wiki/Control-module](url)).

As of my knowledge, I have made all the necessary configuration specified from the README files. I am trying to use the RECORD sub-module from the HTTP module.

As the README suggests

`curl "http://localhost:8080/control/record/start?app=myapp&name=mystream&rec=rec1"
`
`curl "http://localhost:8080/control/record/stop?app=myapp&name=mystream&rec=rec1"
`
the above commands as examples to start/stop recording the stream.

In my case, after starting picam with
`./picam --alsadev hw:1,0 --tcpout tcp://127.0.0.1:8181
`

I am using:
`curl -i "http://localhost:8080/control/record/start?app=webcam&name=mystream&rec=rec1"
`

The output shows as:
> root@raspberrypi:/home/pi/picam# curl -i "http://localhost:8080/control/record/start?app=webcam&name=mystream&rec=rec1"
HTTP/1.1 200 OK
Server: nginx/1.15.8
Date: Fri, 25 Jan 2019 16:40:14 GMT
Content-Length: 35
Connection: keep-alive
/tmp/rec/mystream-1548434414all.flv

But I couldn't find `/tmp/rec/mystream-1548434414all.flv` file.

Please help, what am I doing wrong..

viveksanagari, 2019-01-25:
Update: I have forgotten to mention that I am using them with out-of-the-box picam-streamer ([https://github.com/iizukanao/picam-streamer](url)) solution and need help to fix the problem mentioned in the above comment. Thank you for any help in advance.

iizukanao, 2019-01-28:
@viveksanagari Do you see any error message in your nginx log file?

viveksanagari, 2019-01-28:
Hi @iizukanao 
Thank you very much for responding.

I have these log files for nginx:
`/usr/local/nginx/logs/access.log`
`/usr/local/nginx/logs/error.log`

When I use this command,
`curl -i "http://localhost:8080/control/record/start?app=webcam&name=mystream&rec=rec1"`

The output shows as:

> root@raspberrypi:/home/pi/picam# curl -i "http://localhost:8080/control/record/start?app=webcam&name=mystream&rec=rec1"
HTTP/1.1 200 OK
Server: nginx/1.15.8
Date: Fri, 25 Jan 2019 16:40:14 GMT
Content-Length: 35
Connection: keep-alive
tmp/rec/mystream-1548669298all.flv

There is nothing logged into **/usr/local/nginx/logs/error.log**. 
However, **/usr/local/nginx/logs/access.log** log has a new entry as below.
`127.0.0.1 - - [28/Jan/2019:09:51:02 +0000] "GET /control/record/start?app=webcam&name=mystream&rec=rec1 HTTP/1.1" 200 35 "-" "curl/7.52.1"`

iizukanao, 2019-01-28:
@viveksanagari Thanks for the feedback. Can you also make sure that you can record other stream that is not produced by picam without problem?

iizukanao, 2019-01-29:
@viveksanagari I don't know what that 204 response means. Can you record OBS stream without any problem?

viveksanagari, 2019-01-29:
Hi @iizukanao 
I was not able to record from OBS. I am not sure it is picam, or it is me doing something wrong.

iizukanao, 2019-01-31:
@viveksanagari Please make sure that you understand how to correctly use nginx-rtmp-module first. Unfortunately, I don't have time to look into this problem further right now without more detailed info. Please attach Wireshark packet capture if you can.