iizukanao/picam #123: error: cannot set sample format (Invalid argument)

jianlinwei, 2019-02-25:
I have a sound card from RASPIAUDIO.COM 
Audio DAC HAT Sound Card (AUDIO+SPEAKER+MIC) for Raspberry Pi
I can use pyaudio to record audio and use picamera to record video.

Here are the commands:
**pi@raspberrypi:~/picam $ arecord -l**
```
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

**pi@raspberrypi:~/picam $ ./picam**
```
configuring devices
error: cannot set sample format (Invalid argument)

```

Thanks.

jianlinwei, 2019-03-09:
I downloaded arecord source code and debug it. I found arecord use "default" as device name. But picam use hw:0,0 as device name by default. If use hw:0,0, arecord will report the error too.
So I used default as device name in picam. It works well.

I don't know why.

akashmiami, 2020-05-24:
hi there, I am having the same issue. How did you solve this? did you say: ./picam --alsadev default ? Because I tried that approach and it did not work. Just curious. Thanks.

lucasrangit, 2021-01-09:
@akashmiami what does `arecord --dump-hw-params -D hw:0,0` return for you?

In my case, 

```
$ arecord --dump-hw-params -D hw:0,0
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:0,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 192000]
PERIOD_TIME: (166 8192000]
PERIOD_SIZE: [32 65536]
PERIOD_BYTES: [256 524288]
PERIODS: [2 4096]
BUFFER_TIME: (333 16384000]
BUFFER_SIZE: [64 131072]
BUFFER_BYTES: [256 524288]
TICK_TIME: ALL
--------------------
arecord: set_params:1339: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE
```

MiWeber, 2021-02-01:
@lucasrangit I've got the same error...
`arecord -l` says, that the usb microphone is connected to hw:1,0 or hw:CARD=AUDIO,DEV=0
when I try to execute `./picam --alsadev hw:1,0` I receive the error `cannot set sample format`
when I execute `./picam --alsadev hw:1,0 -r 192000`, I receive the error `[libfdk_aac @ 0x239d8c0] Specified sample rate 192000 is not supported`.

The output for `arecord --dump-hw-params -D hw:1,0` is:
```
$ arecord --dump-hw-params -D hw:1,0
Aufnahme: WAVE 'stdin' : Unsigned 8 bit, Rate: 8000 Hz, mono
HW Params of device "hw:1,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S24_3LE
SUBFORMAT:  STD
SAMPLE_BITS: 24
FRAME_BITS: 24
CHANNELS: 1
RATE: 192000
PERIOD_TIME: [1000 910219)
PERIOD_SIZE: [192 174762]
PERIOD_BYTES: [576 524286]
PERIODS: [2 1024]
BUFFER_TIME: [2000 1820443)
BUFFER_SIZE: [384 349525]
BUFFER_BYTES: [1152 1048575]
TICK_TIME: ALL
--------------------
arecord: set_params:1339: Sample-Format nicht unterstützt
Available formats:
- S24_3LE
```