iizukanao/picam #125: take photo feature

jianlinwei, 2019-03-23:
Hi, iizukanao,

Is it possible to add a take photo feature?
Just like hooks/start_record, add hooks/take_photo to take a photo, the filename can be included in hooks/take_photo.

Thanks.

wasalm, 2019-05-11:
It is not possible in the current version. One alternative you can do is to enable preview mode and make a screenshot with raspi2png(https://github.com/AndrewFromMelbourne/raspi2png)

ADoncel, 2019-05-16:
Hello, I've been following the library for a long time, I'm going to start adding that configuration to the code due to OpenMAX has already everything for you to use it and I need it as a part of a project I'm developing for my company.

I have this as reference:
OpenMAX™ Integration Layer Application Programming Interface Specification:
http://www.jvcref.com/files/PI/documentation/ilcomponents/

VMCS-X OpenMAX IL Components:
http://www.jvcref.com/files/PI/documentation/ilcomponents/

My plan is to keep everything like it is for the moment and just make another hook, like shot_still or so, to be as simple as possible at the moment, and of course, shot stills either during recording or with the idle camera.

Every advice, recommendation, documentation that could be helpful are welcome.

wasalm, 2019-05-16:
If I were you I, I would try one of the following approaches.

1) Copy the code of the render_camera and alter it to use EGL_render. Then retreive the image of the EGL buffer (Maybe this is interesting: https://github.com/drhastings/cameragl). The upshot of this method is that it fully uses the GPU. However, internally it will copy all the data to an EGL buffer which will slow things down.

2. Picam already received an H.264 video stream. Alternatively you can use ffmpeg's codebase to retreive a single frame. Although it utilizes the CPU, the momory usage will be low.

I don't have any experience with ffmpeg programming so I can't help you

feacluster, 2020-03-23:
I have been experimenting with this php script ( called from cron every minute ). The idea is to extract the last frame of the current video file being recorded. The problem is sometimes it works and other times gives strange errors. Here's the code if anyone has any ideas. Will see if I can post the error messages later:

`/////////////////////////////

function grab_last_frame() {

global $hostname;

$out = shell_exec ( "cat logs/run_picam_output.txt | egrep '\.ts$'" );

if ( preg_match ( '/(rec\/.*?\.ts$)/', $out , $matches ) ) $file = $matches[1];
    else { "Echo no match found for $out\n"; return ; }

$file = preg_replace ( '/archive/', 'tmp', $file );

echo $file;

// shell_exec ( "ffmpeg -y -sseof -3 -i $file -update 1 -vf scale=640:480 -q:v 1 /var/www/html/share/$hostname.jpg" );

} // end function
`

bmatusiak, 2020-08-25:
+1 for  `hooks/take_photo`