iizukanao/picam #128: warning: Record buffer is starving.

dangentile, 2019-07-11:
Running the following setup:
Raspberry Pi 3 Model B Plus Rev 1.3
Pi Firmware: Jun 20 2019 16:14:44 
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start_x)
Kernel: Linux 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux
Picam 1.4.7-binary

I am getting the error in picam.log
warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.

Killing the process and restarting is not always effective to bring back the stream. Most times removing the power for several minutes is required. The issue is sporadic. Sometimes I can run for several days without issue, other times only a matter of hours. Lately it has been the latter. The device ran flawlessly for a couple of months and then this started. I rebuilt the whole system on Buster and the issue continues.

When picam crashes, the pi and other services seem to be operating normally. nginx webserver is the only other heavily used process and it continues to serve it's pages without issue.

Any advice on how to make this more stable, it seems like this issue was resolved in this build? Any more logs I could be checking to find out more about the cause?

iizukanao, 2019-07-14:
Thanks for your report. Looks like picam fails to capture video when the room is dark. I'm looking into this problem.

iizukanao, 2019-07-14:
It possibly relates to encoder buffer size and ambient lighting. When the room is almost complete dark, camera produces weird pinkish image (this also happens with raspivid), and it generates large difference between two frames which exceeds encoder buffer size. I'm still looking into this problem, but you can mitigate the issue by using `--autoex` option for now.

dangentile, 2019-07-15:
Interesting, and thank you for looking into this. I'm using as a baby monitor and it has been working very well. Great work, thanks for putting it out there!

I will give --autoex a try and report back!

I'm still confused as to why a simple restart of the service, and a reboot were not always able to bring it back into operation. I'm using the /run/shm/ RAM drive setup, Should that be cleared out when the service is killed, or the pi reboots?

Let me know if can I send you any logs to help with the bug.

Thanks again!

iizukanao, 2019-07-15:
When picam is unresponsive, you may have to send `kill -9` (SIGKILL) in order to terminate picam process. You don't have to clear out /run/shm/.

Just for reference, can you let me know picam command line arguments you are using?

dangentile, 2019-07-15:
Here is my command line - I was going to reply later, but I have already been using the --autoex so that may not be a viable workaround

picam -o /run/shm/hls --samplerate 44100 --channels 1 --audiobitrate 96000 --videobitrate 4000000 --vfr --avclevel 3.1 --autoex --time --alsadev hw:1,0 >/var/log/picam.log 2>&1

I'm stating picam at boot by calling a shell script from /etc/rc.local if that matters

iizukanao, 2019-07-15:
Thanks for your quick response. I will publish bugfix tomorrow.

closed 2019-07-16 by iizukanao

iizukanao, 2019-07-16:
@dangentile I've fixed this bug in the latest master branch. I'll publish new binary release (v1.4.8) later. Closing this issue for now, but please feel free to file a new issue if you are still having the problem.

iizukanao, 2019-07-16:
@dangentile Here is picam binary v1.4.8. Please see if it works. https://github.com/iizukanao/picam/releases/tag/v1.4.8
Thank you again for reporting this!

dangentile, 2019-07-16:
Happy to help!

v1.4.8 is operational. I will report back in a few days if there any new or further issues.

Thanks!

dangentile, 2019-07-23:
Hi there, just writing to say the system has been running for 7 days without issue. Thanks again for the help and great software!

iizukanao, 2019-07-24:
@dangentile Thank you for letting me know!

harryjubb, 2020-07-26:
@iizukanao @dangentile 

Having this issue currently, with no clear reason as to why it's started happening.

Had streaming working with no problems until it apparently randomly started occuring, with no changes as far as we can tell apart from moving the Pi/camera to a lighter environment.

Currently, the stream will work for typically a few seconds before cutting off whenever the `warning: Record buffer is starving...` error hits.

Have tried adding the `--autoex` option with no success.

Currently running as below:

https://github.com/harryjubb/bee_iot/blob/a07fe8263487cd29a2526135e6d5cf4263355139/hive/services/av_streaming/picam/entrypoint.sh#L15

Any help/ideas appreciated!

feacluster, 2020-08-17:
Just fyi, your installation instructions still reference v1.4.7. See:

# Install picam binary
wget https://github.com/iizukanao/picam/releases/download/v1.4.7/picam-1.4.7-binary.tar.xz
tar xvf picam-1.4.7-binary.tar.xz
cp picam-1.4.7-binary/picam ~/picam/

feacluster, 2020-08-30:
I upgraded to the new v1.4.9 but still occasionally get the message:

warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.

It seems when this message appears the camera stops recording and the video file gets corrupted ( youtube can't process it ).

This is the command I am using:

~/picam/picam --alsadev hw:1,0 --vfr --maxfps 30 --minfps 10 --shutter 20000 -v 1000000 --mode 2 -w 1640 -h 1232

feacluster, 2020-11-23:
Update: I've increased the GPU memory under "raspi-config" from 128 to 256. Now it us running A LOT more stable. Have only seen this "starving buffer" message only once in a 100+ video recordings . Before I was getting it one in every 20 videos.

feacluster, 2021-01-27:
Another update: Ignore my previous message. Even after increasing the GPU memory I still get random video corruption due to the starving buffer. Still seems to happen once in 20 videos . Each of my videos is 45 minutes in length.  I suppose the only solution is to decrease my resolution from 1640 X 1232 .

feacluster, 2021-03-15:
Think I finally got to the bottom of this ( famous last words.. ). The problem was using mode 4 with 1642X1232. When I switched to mode 2 with same resolution, the recording became a lot more stable. I have not noticed any starving buffers in hundreds of videos. Before it used to be ~5-10% of all videos. So here's what works:

~/picam/picam --alsadev hw:1,0 --vfr --maxfps 30 --minfps 5 -v 1000000 --mode 4 -w 1640 -h 1232 > logs/run_picam_output.txt 2>logs/run_picam_error.txt &

iizukanao, 2021-03-16:
@feacluster Thanks for letting me know!

wasalm, 2021-03-30:
I've got the same problem for a camera inside a birdhouse. However I noticed that now the sun is shining and the birds are building their nest, there is much more detail to capture. I had a bitrate of 5M, but now I reduced the bitrate to 1.5M it now works again.

commit 8058c178 (2019-07-16): Fix hangup when video_encode produces large frame

video_encode hangs up when there is a large difference in video images
between two adjacent frames and all OMX buffers get used up. Increasing
the number of buffers solves this problem.

Fixes #128