iizukanao/picam #129: good job

fagol, 2019-10-12:
Hello, sorry for my english, i'm french.

Thanks for your work. A smal detail, this does not work, even changing the N° of release, i take it manually:  
# Installer le binaire Picam
wget https://github.com/iizukanao/picam/releases/download/v1.4.7/picam-1.4.7-binary.tar.xz
tar xvf picam-1.4.7-binary.tar.xz
cp picam-1.4.7-binaire / picam ~ / picam /

My first question : is it possible to work it with python projet, and how can do start,or stop in python.
My second an last : is it possible to change the way of record, like USB HDD?

Thanks a lot

iizukanao, 2019-10-14:
1) Yes, you can start/stop picam with a Python script. Please refer to [subprocess](https://docs.python.org/3/library/subprocess.html) documentation for how to execute commands.
2) Yes, mount your HDD, and change "rec" directory in picam to a symbolic link to your HDD mountpoint.

fagol, 2019-10-15:
Tanks for your come back it's work good for me.

iizukanao, 2019-10-15:
Thanks for your feedback. I'm closing this issue, but please feel free to open a new issue when you encounter a problem.

closed 2019-10-15 by iizukanao

feacluster, 2020-01-02:
If anyone is interested, I have some example code to control picam from web-based php script that calls a bash script.