iizukanao/picam #132: Dropped frames every 15-30 seconds

feacluster, 2020-01-02:
I am using a raspberry pi 3B with picam 1.4.7. Seems every 15-30 seconds I experience dropped frames. The video goes still for a split second then jumps ahead. I have tried both these commands with picam:

picam --alsadev hw:1,0 
picam --alsadev hw:1,0 --vfr --maxfps 30 --minfps 10 --shutter 20000 -v 1000000 -w 1024 -h 768

I need to record upto 60 minutes, so can not use the /dev/shm folder. These are being written to the micro SD card. 

You can watch an example of it here:

https://www.youtube.com/watch?v=REf9hh1kklo

Is there some gpu setting I need to play around with?

feacluster, 2020-01-06:
Update: I tried playing around with --mode and it seems to give fewer dropped frames with some modes.. Mode 7 seems best but the field of view is smaller and seems the quality is not so sharp..
Any other suggestions to try?

iizukanao, 2020-01-06:
Not utilizing /dev/shm might be the cause of such issue. Can you add `--verbose` option to your picam command then provide me a full output?

feacluster, 2020-01-09:
Thanks, I switched it back to /dev/shm and now the dropped frames issue has mostly gone away . Initially I thought using /dev/shm would limit the video size to the size of /dev/shm . But seems that is not the case? Looks like picam writes the video file back to the SD card so the size of the SD card is the limiting factor. Is that correct?

iizukanao, 2020-01-12:
@feacluster Sorry for the late reply. Yes, you're correct. /dev/shm is used only for write buffering (reduces the frequency of writes to SD card).

feacluster, 2020-03-21:
Just another update. I switched to a different SD card and now experience zero dropped frames. This is the one I used:

Samsung (MB-ME32GA/AM) 32GB 95MB/s (U1) microSDHC EVO Select Memory Card

closed 2020-03-21 by feacluster