iizukanao/picam #133: After a few minutes framerate drops very low but CPU is fine

JamesSwift, 2020-01-24:
Firstly, thank you so much for this amazing project! It's been really helpful with my baby monitor and a few other cameras.

I have a pi zero with a usb microphone. On that particular pi when I start picam it works amazingly, but after a few minutes, suddenly the perceived framerate drops to almost 2 fps. The cpu is running around half and the memory usage is around half too so there doesn't seem to be a leak. The video will continue indefinitely (tested for round 20 hours so far), but it never gets back to the smooth frame-rate it started with. I'm having no joy in tracking this down. Any ideas?

iizukanao, 2020-01-25:
Thanks for using picam.
What command line options are you using? You can use `-f <fps>` option in order to set a fixed frame rate, like `-f 30`.

JamesSwift, 2020-01-25:
Thanks for the quick reply! I'm running:

`picam -o /run/shm/hls --autoex --time --rotation 180 -w 1080 -h 810 -v 5000000 --volume 20 --alsadev hw:1,0 `

I've come up with a working theory. I use the `autoex` flag, and the cameras are in low light (baby's room). When I increase the light the perceived frame-rate increases. So I think the exposure is being boosted to maybe 1/3 second by autoex during darker periods. What's weird though is that if I restart picam it starts with a short exposure then after a few minutes it goes back to the long exposure.

iizukanao, 2020-01-26:
You're right, `--autoex` automatically enables `--vfr` (variable frame rate) mode. Don't use `--autoex` and use `--fps <fps>` instead if you want a fixed frame rate. When `--autoex` is used, picam starts with short exposure first.