iizukanao/picam #135: [feature request] Minimum volume audio cutoff

HWiese1980, 2020-03-15:
First off, awesome program! Works like a charm on my Pi. 

What I'm looking for, however, is to have some sort of cutoff threshold for very low noise. I use it in a babyphone setup and would like to not transmit any audio that's lower than a crying or talking baby. The buzzing background noise does not have to be transmitted and can remain silent. So, only audio above a certain level should be transmitted at all. I've already taken a look at your code and I suppose that to add this feature, where the volume factor is incorporated would be a good place to start. I'm just not sure about the numbers, what `audio_min_volume` and `audio_max_volume` do and if they can be used for this purpose. 

I would do it myself but I've got trouble building picam on my Pi. Probably nothing related to picam, but to my build environment. So no need for support here. Just wanted to request the feature. 

Keep rocking

iizukanao, 2023-02-16:
This feature has been implemented as `--ngate` option. (#197)

iizukanao, 2023-02-16:
@HWiese1980 Thank you for your request. I apologize for the long period of time that has passed without a reply.

commit 8bb943a5 (2023-02-16): Merge pull request #197 from francescovannini/noisegate

Feature: Noise Gate. Mute sound when below a threshold.

Fixes #135
closed 2023-02-16 by iizukanao