iizukanao/picam #136: /run/shm/ symbolic links randomnly get deleted?

feacluster, 2020-03-21:
I created a cron script to automatically start my camera when certain conditions are met. I am running into an issue where some recordings randomly get killed. It seems the symbolic links to /run/shm are broken when this happens. I did some research and seems this maybe what is going on:

https://superuser.com/questions/1117764/why-are-the-contents-of-dev-shm-is-being-removed-automatically

The solution is:

Edit /etc/systemd/logind.conf, uncomment the line RemoveIPC=yes, change it to RemoveIPC=no, save, and reboot the system

Before I do this I am wondering if anyone else ran into this issue and how they resolved it?

inseltraeumer, 2020-05-26:
I solved the issue by checking for the existence of one of those symbolic links before starting picam and if it doesn't exist I run that make_dirs.sh script mentioned in the README.md

feacluster, 2020-05-27:
I am not sure if running the make_dirs.sh script before starting picam will fix the issue. I did try that initially but still experienced random terminations. What I have done is changed the state rec hooks and hls folders to root owner and made them 777. Then did chmod 777 as root the shm folder & the archive folder & the symbolic link to archive.

Seems to be stable now. But will need more exhaustive testing.

feacluster, 2020-07-28:
My workaround above has been "mostly stable". Once in a while picam will stop recording because the temporary file it writes to /run/shm/rec/tmp gets deleted while it is recording. It looks like I will need to do the solution of setting RemoveIPC=no as mentioned in my first post. I would still prefer not to do it because of unintended consequences. Anyone have a workaround to the tmp file getting deleted?

feacluster, 2020-08-13:
Just an update I switched to setting "RemoveIPC=no" and now picam has been very stable.

feacluster, 2020-08-13:
Also another tip is to add this line to your crontab:

@reboot /var/www/html/make_dirs.sh

closed 2020-08-13 by feacluster