iizukanao/picam #138: new HQ-cam on rpi4 at 1920x1080 fails (port buffer)

gepatto, 2020-05-08:
I'm getting `error: failed to enable port buffers for video_encode 200` when trying to run picam
with the new HQ-cam on a raspberry pi 4 at 1920x1080 with up to date buster.

Command used is:
`picam --verbose -w 1920 -h 1080 --rotation 180 --tcpout tcp://127.0.0.1:8181`

verbose output:
using AVC High Profile Level 4
warn: fps > 20 might not work properly when width and height is large.
      Use lower --fps or use --vfr. If you still want to use this
      configuration, see if picam keeps up with 25.0 fps using --verbose.
video_width=1920  
video_height=1080  
video_fps=25.000000  
fr_q16=1638400
video_pts_step=3600
video_gop_size=25
video_rotation=180
video_hflip=0
video_vflip=0
video_bitrate=2000000
video_avc_profile=high
video_avc_level=4
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
alsa_dev=hw:0,0
audio_channels=1
audio_sample_rate=48000
audio_bitrate=40000
audio_volume_multiply=1.000000
is_hlsout_enabled=0
is_hls_encryption_enabled=0
hls_keyframes_per_segment=1
hls_number_of_segments=3
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=1
tcp_output_dest=tcp://127.0.0.1:8181
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
is_timestamp_enabled=1
timestamp_format=%a %d. %b %Y  %H:%M:%S
timestamp_layout=15
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
timestamp_pos_x=0
timestamp_pos_y=0
timestamp_text_align=1
timestamp_font_name=FreeMono:style=Bold
timestamp_font_file=
timestamp_font_face_index=(null)
timestamp_font_points=28.000000
timestamp_font_dpi=96
timestamp_color=ffffff
timestamp_stroke_color=000000
timestamp_stroke_width=1
timestamp_letter_spacing=0
is_preview_enabled=0
is_previewrect_enabled=0
preview_x=0
preview_y=0
preview_width=0
preview_height=0
preview_opacity=255
blank_background_color=0x0
is_audio_preview_enabled=0
audio_preview_dev=plughw:0,0
record_buffer_keyframes=5
state_dir=state
hooks_dir=hooks
configuring devices
exposure control: auto
setting exposure:
  eMetering: 0
  xEVCompensation: 0
  nApertureFNumber: 0
  bAutoAperture: 0
  nShutterSpeedMsec: 0
  bAutoShutterSpeed: 1
  nSensitivity: 0
  bAutoSensitivity: 1
Set video_encode state to idle
Enable port buffers for camera 71
Enable port buffers for video_encode 200
error: failed to enable port buffers for video_encode 200

gepatto, 2020-05-08:
Ok, I think I may have solved it. 
It seems a 128mb Memory-split is not enough. 
Raised it to 160mb and now it's working..

closed 2020-05-08 by gepatto

iizukanao, 2020-05-08:
@gepatto Thank you for the info!
I still haven't got an HQ cam yet. Will add that info to the README!