iizukanao/picam #139: Freezes

ouafnico, 2020-05-14:
Hi

I'm using the last version of picam (1.4.8).

When I play the video recorded, I got a lot of freezes on the videos, but the sound continue correctly.

The picam doesn't say anything wrong:

using AVC High Profile Level 4
configuring devices
capturing started
o-o-o+o+o-o-o-o+o-o-o+o-o-o-disk_usage=3% start rec to rec/archive/2020-05-13_21-56-40.ts

I'm using it like this:

./picam -w 1920 -h 1080 -v 2000000 --fps 20 --volume 20 --alsadev hw:1,0

I don't think it's a sdcard issue, the file is very light, and it's a class 10.

Do you have an idea?

Thanks

lightray22, 2020-05-19:
I also have this issue.  I found that lowering the resolution to 720p got rid of the freezes.  Another work-around is to keep it ta 1080p but enable VFR.  For me, this got rid of the freezes but the average frame rate according to VLC was in the low 20's, not the 30 I set it at.  It's almost like picam is having issues keeping up somehow, though I know the Pi's h264 encoder is capable of 1080p30 (works with with raspivid) and picam's CPU usage is only around 60%.

lenzp, 2020-08-26:
Same to me. I set video size to 1024x720 and frame rate to 25. If I play the video with Media Player Classic Home Cinema, it freezes extremely, and when I click on "properties", it shows me:
Video: MPEG4 Video (H264) 1024x720 90090.094fps [V: h264 constrained baseline L3.1, yuv420p, 1024x720]
Audio: AAC 48000Hz mono 37kbps [A: aac lc, 48000 Hz, mono, 37 kb/s]

BUT: if I insert this video into the timeline of a project in Adobe Premiere, everything looks fine, and nothing is frozen. The frame rate is shown as a little bit below 25.

But those results came from some experiments. I formatted a partition on my SD card as NTFS, and this seems to be a very bad idea because it takes much more time to save data.

So I went back to FAT32, left all those video size and fps settings away, and now I get videos in 1280x720 more than 25 fps. I was surprised by the fact that there is no delay between audio and video.

coeli, 2020-08-28:
Same issue with raspberry pi 4 4Gb and camera_HQ
256mb reserved to GPU

Video Freeze at regular time, every 2-3s 

./picam -w 1280 -h 720 --fps 25 --> NOK
./picam -w 1920 -h 1080 --fps 20  --> OK
./picam -w 1920 -h 1080 --fps 25 --> NOK

iizukanao, 2020-08-30:
fps higher than 20 may be too high for resolutions larger than 1280x720 because picam does some processing in addition to H.264 encoding. Try lower fps, resolution, or `--vfr` option. You can go 30 fps for 1280x720.

lenzp, 2020-08-30:
IMHO it is rather useless to work with large resolutions and high fps because you can not expect really good quality when the video bitrate is 2000000. Fast movements in the pictures lead to blocks and artefacts in the pictures. Recording video and audio is a difficult task. Somewhere I read the statement that Raspberry is in general too weak to capture audio from USB and compress it with AAC, and so it seems to be a miracle that picam is even able to record video on the same time.

From my experiences it is better to reduce resolution and fps, but enhance the video bitrate. This leads to "starving buffer" errors. Using --vfr seems not to be very useful, it leads to videos which have wrong speed in some players and also in Adobe Premiere CS6. Best results come from lowering --gopsize. I recommend
picam --width 960 --height 720 --gopsize 15 --fps 25 --videobitrate 4000000 --samplerate 22050 --audiobitrate 20000 --alsadev hw:2,0 &
which leads to good results and a file size of ~2 GB per hour. 

One question: if file size does not matter, what is the smallest possible gopsize which needs the minimum of calculation power?

coeli, 2020-08-31:
Idea for new release, user can be enable/disable backend processing options to preserve cpu load for FullHD resolution and 25 fps

feacluster, 2020-09-27:
> From my experiences it is better to reduce resolution and fps, but enhance the video bitrate. This leads to "starving buffer" errors. Using --vfr seems not to be very useful, it leads to videos [...]

Thanks for these tips! I too would randomly get the "Record buffer is starving." error messages which would lead to corrupted video file.  I am going to test your settings and see if that resolves it.

Curious how to get 60fps? Sounds like it is impossible with the picam?

feacluster, 2020-11-23:
Update: I've increased the GPU memory under "raspi-config" from 128 to 256. Now it us running A LOT more stable. Have only seen this "starving buffer" message only once in a 100+ video recordings . Before I was getting it one in every 20 videos.