iizukanao/picam #140: ~1 fps Night Exposure Mode and Cannot Record.

summrum, 2020-05-26:
Hi, I am using a Raspberry Pi 3B+ with Raspbian Buster, PiCam v 1.4.8, a raspberry pi camera module and adafruit usb microphone. I am exposing the PiCam HLS stream to nginx on my raspberry pi to view from web browsers on various devices. 

The stream works fairly well with the camera exposure set to day mode, but setting to exposure night mode drops the framerate to around 1 fps (unusable). Setting a minfps value does not help the issue at all. This also happens when autoex is set and the camera drops to night exposure mode.

Secondly, I am completely unable to record. I have tried both using the RAM drive and without, but creating the hooks/start_record file and stop_record file does not produce any recording no matter whether using the RAM disk or not.

Thank you.

inseltraeumer, 2020-05-26:
Concerning the exposure mode, you could use "nightpreview" instead. That one gives me around 18 FPS. 

There are also "no IR" camera modules (dubbed night or low light vision") for the Raspberry Pi, if you really do nightly recordings. 

The reason for the low FPS with low light exposure settings is that to achieve a brighter recording, the camera sensor has to collect more light for each picture/frame which means the FPS goes down. The "night" setting has a very long exposure and thus also a very low FPS. 

Comcerning the other issue you mention that you can't record, I'm not sure I understand you. I thought you were also having problems with low FPS, and wouldn't that mean that recording does work?

summrum, 2020-05-27:
Hi, thank you for your reply. I should have been more clear, sorry. The inability to record is a separate issue; no matter the exposure setting, bitrate, use of RAM disk etc. creating the hooks/start_record and stop_record files do not give me any recordings. There is nothing saved in the rec/archive folder. I have also searched the pi SD card and not found any recordings anywhere. The low fps is what I judge from my observation when viewing the stream, I have not measured it exactly by opening any files in software.

inseltraeumer, 2020-05-27:
When I use Picam with the --tcpout option, it doesn't save any recordings for me either. I guess that's intententional - or just unsupported. I guess you could try to use ffmpeg to capture the Picam output and then forward it to a URL and save it at the same time. 

http://trac.ffmpeg.org/ticket/1382

Here's part of a script I use to stream to YouTube, I guess you could tweak that. 

```FFREPORT=file="ffreport-$(date).log":level=32 /home/pi/ffmpeg/ffmpeg \
  -re \
  -i tcp:127.0.0.1:8181?listen \
  -filter:a aresample=async=1000 \
  -c:a aac \
  -c:v copy \
  -f flv rtmp://a.rtmp.youtube.com/live2/$1 &

./picam \
  --alsadev plughw:1,0 \
   -c $audio_channels \
   -r $audio_framerate \
   -a $audio_bitrate \
  --time \
  --timeformat "$timeformat" \
  --hflip \
  --vflip \
   -w $vwidth \
   -h $vheight \
  --ex $exposure \
  --tcpout tcp://127.0.0.1:8181

touch hooks/start_record```

summrum, 2020-05-28:
Thanks again for the reply; I hadn't considered that the HLS stream was the issue. Will try capturing the stream with ffmpeg as you suggest.

summrum, 2020-05-28:
Just an update, this works well for me. Thanks again. I run the following command to record the HLS stream:

ffmpeg -i "http://127.0.0.1/hls/index.m3u8" -c copy -bsf:a aac_adtstoasc "rec_`date +%s`.mkv"