iizukanao/picam #141: stream not showing until exiting picam

joop339, 2020-06-01:
So this is gonna be a hard one to explain.
Following the instructions and starting a live stream i get no errors.
I open vlc and start the video stream however i get neither sound nor video.

But when i close the stream using ctrl-c in the terminal a short clip suddenly show up in vlc. This happens everytime i exit.

some stuff i figured out that might be related.

1.
When performing a reboot i cant run ./picam and get this error: "error creating state dir (./state): File exists" 
I can fix this by running make_dirs.sh.

2.
i cant run make_dirs.sh without sudo i get this error: "ln: failed to create symbolic link '/run/shm/rec/archive': Permission denied"

3.
i cant run ./picam without sudo i get this error: "stat directory: Permission denied
State file open failed: Permission denied"

i hope you can help me with this. If there's any more information you need please send me a message.

joop339, 2020-06-01:
[debug.log](https://github.com/iizukanao/picam/files/4713260/debug.log)

also here is my debug log from vlc. I don't see anything really weird in there, but then again iam not that experienced.

kendaw, 2020-06-06:
I'm having this exact same problem. Installed everything today using the Raspberry Pi OS (32-bit) Lite (27 May 2020).

kendaw, 2020-06-06:
@joop339 any luck in the past few days in resolving this issue?

joop339, 2020-06-06:
@kendaw sadly no. I went back to streaming using VLC directly.

kendaw, 2020-06-06:
@joop339 from the host/client itself, I assume?

kendaw, 2020-06-19:
@joop339 if you still want to view the feed from a remote client, I was able to using this web player: https://kamranicus.com/guides/raspberry-pi-3-baby-monitor#optionally-adding-a-web-player

SebastianSTARITZ, 2020-09-10:
I do have the same issue. 

I'm using buster lite and 1.4.9

- i tried -p (preview) and capturing was OK
- .ts files ar generated and accessible via client. Each is ~ 149kB big.
- i downloaded one .ts file and tried to open it "offline" via VLC -> still no video 
- i tried different versions 1.4.7 , 1.4.8, 1.4.9 binaries ..... still no change
- i tried different start parameters like changing fps, widht, height, avclevel and so on -> no change

no error in /var/log/picam
no suspicious output in /var/log/messages when greping for "cam", "picam", "video", "hls", "avc", "codec"

webplayer from kamranicus.com is working OK

any ideas where to look at in order to debug?

SebastianSTARITZ, 2020-09-10:
> 
> 
> So this is gonna be a hard one to explain.
> Following the instructions and starting a live stream i get no errors.
> I open vlc and start the video stream however i get neither sound nor video.
> 
> But when i close the stream using ctrl-c in the terminal a short clip suddenly show up in vlc. This happens everytime i exit.
> 
> some stuff i figured out that might be related.
> 
> 
> When performing a reboot i cant run ./picam and get this error: "error creating state dir (./state): File exists"
> I can fix this by running make_dirs.sh.
> 
> 
> i cant run make_dirs.sh without sudo i get this error: "ln: failed to create symbolic link '/run/shm/rec/archive': Permission denied"
> 
> 
> i cant run ./picam without sudo i get this error: "stat directory: Permission denied
> State file open failed: Permission denied"
> and
> avio_open for /run/shm/hls/1.ts failed: Permission denied
> 
> i hope you can help me with this. If there's any more information you need please send me a message.

Hi, i checked your log. It seems to point out some issues with the .ts files. At least, that's what i suspect after trying some things 

`ts warning: Broken stream: pid 256 sends packets with dts 888900us later than pcr, applying delay
ts debug: first packet for pid=257 cc=0x5
adaptive debug: Retrieving http://192.168.178.115:80/hls/11.ts @0
ts warning: discontinuity received 0x0 instead of 0x8 (pid=17)
ts error: libdvbpsi error (PSI decoder): TS discontinuity (received 0, expected 8) for PID 17
ts debug: SDTCallBack called
ts debug: Can't read TS packet at 552720`