iizukanao/picam #144: When i build picam, i got this error.

joelihn, 2020-09-11:
`cc stream.o hooks.o mpegts.o httplivestreaming.o state.o log.o text.o timestamp.o subtitle.o dispmanx.o -o picam -g -Wl,--whole-archive -lilclient -L/opt/vc/lib/ -L/usr/local/lib -lbrcmGLESv2 -lbrcmEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lpthread -lrt -L/opt/vc/src/hello_pi/libs/ilclient -Wl,--no-whole-archive -rdynamic -lm -lcrypto -lasound `pkg-config --libs freetype2` `pkg-config --libs harfbuzz fontconfig libavformat libavcodec`
/usr/bin/ld: /usr/local/lib/libavformat.a(fifo.o): in function `fifo_init':
/home/pi/Joe/picam/ffmpeg/libavformat/fifo.c:519: undefined reference to `__atomic_store_8'
/usr/bin/ld: /usr/local/lib/libavformat.a(fifo.o): in function `fifo_write_trailer':
/home/pi/Joe/picam/ffmpeg/libavformat/fifo.c:624: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: /home/pi/Joe/picam/ffmpeg/libavformat/fifo.c:631: undefined reference to `__atomic_store_8'
/usr/bin/ld: /usr/local/lib/libavformat.a(fifo.o): in function `fifo_thread_write_packet':
/home/pi/Joe/picam/ffmpeg/libavformat/fifo.c:188: undefined reference to `__atomic_fetch_sub_8'
/usr/bin/ld: /usr/local/lib/libavformat.a(fifo.o): in function `fifo_consumer_thread':
/home/pi/Joe/picam/ffmpeg/libavformat/fifo.c:457: undefined reference to `__atomic_load_8'
/usr/bin/ld: /usr/local/lib/libavformat.a(fifo.o): in function `fifo_write_packet':
/home/pi/Joe/picam/ffmpeg/libavformat/fifo.c:597: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make: *** [Makefile:26：picam] 错误 1
`

joelihn, 2020-09-11:
After add -latomic to LDFLAGS, i can build picam.

But when i run the program,  it crashed.

Here is the information:

`pi@raspberrypi:~/Joe/picam/picam $ gdb --args ./picam -o /run/shm/hls --hlsenc --hlsenckeyuri enc.key   --hlsenckey f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff   --hlsenciv 000102030405060708090a0b0c0d0e0f --alsadev hw:0,0
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./picam...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/pi/Joe/picam/picam/picam -o /run/shm/hls --hlsenc --hlsenckeyuri enc.key --hlsenckey f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff --hlsenciv 000102030405060708090a0b0c0d0e0f --alsadev hw:0,0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x76197420 (LWP 1624)]
configuring devices
[New Thread 0x757ff420 (LWP 1625)]
[New Thread 0x74dff420 (LWP 1626)]
[New Thread 0x745fe420 (LWP 1627)]
[New Thread 0x73dfd420 (LWP 1628)]
[New Thread 0x735fc420 (LWP 1629)]
[New Thread 0x72dfb420 (LWP 1630)]
capturing started

Thread 8 "ILCS_HOST" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x72dfb420 (LWP 1630)]
0x000ce98c in hls_write_packet ()
(gdb) bt
#0  0x000ce98c in hls_write_packet ()
Backtrace stopped: Cannot access memory at address 0x40
(gdb) where
#0  0x000ce98c in hls_write_packet ()
Backtrace stopped: Cannot access memory at address 0x40
(gdb) `

marler8997, 2021-01-22:
I just tried to build and got the exact same backtrace.  My guess is it's a problem with the ffmpeg version I built?    The docs for `BUILDING.md` don't specify a version and/or commit to build from.  Could be a red herring but it seems like the most likely culprit at the moment.  I'll try to update you if I learn more.

marler8997, 2021-01-22:
This PR should fix the segfault: https://github.com/iizukanao/picam/pull/153

You can get around the requirement for `-latomic` by using ffmpeg version `4.3`.  We should update `BUILDING.md` to reflect what version of ffmpeg to pull.

marler8997, 2021-01-22:
And this PR should clarify what version of ffmpeg to build to you don't get the `undefined reference` errors for missing `-latomic`: https://github.com/iizukanao/picam/pull/155