iizukanao/picam #145: Sync Issue - Audio delay 1.1 seconds

sbkomurcu, 2020-10-24:
Hi, first of all, thanks for the Picam project! I have never been able to get the audio to sync up ( apart from static delay) with the video. Thanks to the Picam, this just works amazing! 

However, the audio record starts ~1.1 seconds later than the video. Therefore video and audio are not synced unless I modify the output by shifting audio 1.1 seconds at the outputting. 

Thus, I would like to ask you if it is possible to add ~1.1 seconds delay for the video record to initiate? Or what other solution would you suggest?

Regards,

iizukanao, 2020-10-25:
Hi, thanks for using picam!
You might want to check your ALSA configuration. What's your ALSA configuration for the USB microphone? You should use a device name that starts with `hw:` like `hw:1,0`.

sbkomurcu, 2020-10-25:
Hello! I am using i2s mems mic (sph0645) with the following configuration; "./picam -w 1920 -h 1080 -f 30 --alsadev plug:dmic_hw --volume 5".  When I use "hw:1,0" capturing started without error but there is also no sound at the output. Therefore I used the command ' --alsadev plug:dmic_hw' in which sound is recorded perfectly but must be shifted1.1 seconds.

iizukanao, 2020-10-25:
Hmm, I'm not sure why it introduces 1.1 seconds of delay. Have you tried `plughw:<num>`? You can find the `<num>` by running `arecord -l`.

sbkomurcu, 2020-10-29:
Thanks for the suggestion! Unfortunately, I was not able to fix the issue based on your suggestion. Use of `plughw:<num>` _(i.e. `plughw:0,0`)_ lead me to an issue which is discussed in #86.   On the other hand, the use of `plug:dmic_sv` leads me to the same issue mentioned in #115.

Therefore I am only able to get a proper audio record by using `plug:dmic_hw`, but with a sync issue. 
Since the amount of delay is static throughout the whole record, I was able to fix the issue by doing a workaround during the conversion from `.ts `to `.mp4` by following the command below;

`ffmpeg -i [FILE_NAME].ts -itsoffset -0.4 -i [FILE_NAME].ts -map 0:0 -map 1:1  -acodec copy -vcodec copy synced.mp4`
where `-itsoffset <NUM>` is used to adjust the shifting of the audio track during the conversion process.
 
Regards!

brainrecall, 2022-12-15:
Hmm, I have the same issue, but I'm streaming and not recording. I had to setup a virtual alsa device to merge channels to workaround the slow audio (merge two channels into one). Are there other alsa settings that you know of that could be tweaked?