iizukanao/picam #148: Audio Noise Problem

bill-connelly, 2020-12-28:
Whether I stream the audio, or record a video directly I get the weird noise, pulsing at 1 Hz (which you can't really hear in the sample below). The microphone is fine when I test it on other PCs. 

I tried with various different sample rates apart from the default  (e.g. --samplerate 44100 ) and they never work.

I uploaded a sample here just in case that helps anyone. https://youtu.be/J2GmQLPIvHs

And this is in a perfectly quiet room. There is no fan on the Pi or in the same room.

No errors when I run in --verbose. 

Not a very helpful bug report, but it's really annoying. Let me know if I can run any other tests. Thanks.

lucasrangit, 2021-01-21:
Do you get the same noise when using `arecord`?

iizukanao, 2022-07-07:
Closing due to no response from the issue reporter

closed 2022-07-07 by iizukanao