iizukanao/picam #149: FullHD Stream to Youtube

bostiko, 2021-01-02:
Hallo,
I am trying to making live stream in FullHD to youtube with this SW, but I am not able to set it to reasonable FPS, the motion on the video is snatchy, the FPS is around the 20-24FPS at best. Can you help me to tweak the parameters?
Thank you, Ondrej

My script:
```
url="rtmp://a.rtmp.youtube.com/live2/xxx"
bitrate=192000
samplerate=48000
width=1920
height=1080
log_ffmpeg="/var/log/ffmpeg.log"
log_picam="/var/log/picam.log"

ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a aac -ar $samplerate -ab $bitrate -af "pan=stereo|c0=c1|c1=c1" -f flv $url 2>> $log_ffmpeg &
sleep 1
/home/pi/picam/picam -w $width -h $height --alsadev hw:2,0 --volume 2 -r $samplerate -a $bitrate --avclevel 4.2 --verbose --tcpout tcp://127.0.0.1:8181 >> $log_picam
```

iizukanao, 2021-02-04:
Around 20 fps is the maximum for 1920x1080. Please use 1280x720 for 30 fps.

bostiko, 2021-02-24:
@iizukanao Thank you for your answer. Is there any workaround? Choosing right camera hardware mode?

Or what about repeat some frames programatically? My suggestion is to repeat last frame after every 2 frames. This could increase the FPS from 20 to 30 without people to notice. And in picam, add command-line argument to enable this feature.

My idea is based on my observation of Youtube behaviour. When I send 20 FPS, Youtube displays (I think) 40 frames at the speed of 30FP and then pause a video for 20 frames. And then continue with next data. When somebody moves in the video, the movement is snatchy.

iizukanao, 2022-07-07:
Sorry for the late reply. Publishing 1920x1080@30fps to YouTube is now possible with picam version 2.0.0.

closed 2022-07-07 by iizukanao