iizukanao/picam #158: Text overlay (subtitle) with HLS

neon29, 2021-03-05:
To add subtitle I have proceeded as follows:

Modify my `run_picam.sh`:

```
sudo bash /home/pi/make_dirs.sh
sudo touch /home/pi/picam/hooks/subtitle
sudo cat  /home/pi/picam/sub  > /home/pi/picam/hooks/subtitle
sudo /home/pi/picam/picam -o /run/shm/hls --alsadev hw:1,0
```

where `cat` contains:
```

text=♨☀♻♥⚠
font_file=/home/pi/NotoSansCJKjp-Regular.otf
pt=120
layout_align=middle,center
letter_spacing=40
```

I can then watch and listen to my stream at `http://ip_address/hls/index.m3u8` for example thanks to VLC. However, I do not see any subtitle.

Does subtitles are expected to work with HLS?

iizukanao, 2021-03-16:
@neon29 Yes, subtitles work with HLS. Have you put a font file /home/pi/NotoSansCJKjp-Regular.otf? If not, you can download the file from https://github.com/googlefonts/noto-cjk/raw/main/NotoSansCJKjp-Regular.otf

al3xt3ch, 2021-05-08:
Got the same problem. Tried vlc and webstream via nginx.
`
text=TEST
font_file=/home/pi/Sentinel-Medium.otf
pt=18
layout_align=middle,center

`