iizukanao/picam #159: Reducing flicker with subtitle overlay

Boeeerb, 2021-03-16:
Is there a way of reducing the flicker when updating the text overlay? 

I have a python script saving content every second to a file and moving it to the hook folder, which works great, but the subtitle text on the video often has a flicker when updating - Assuming clearing text before displaying the next.

Really appreciate the work you have in to making this super easy to use project!

iizukanao, 2021-03-16:
@Boeeerb Thank you for your report. Can you share a Python script that demonstrates that problem?

Boeeerb, 2021-03-16:
The main bulk of the python code reads various sensors, and stores in variables that get picked up by these lines;
(Trimmed to the part that writes the file);

```
    if RecordingState == 1:
        f = open("/home/pi/subtitle.txt", "w")
        f.write('''text=System  -  '''+timenow+'''  -  P: '''+str("{:.1f}".format(pressure))+'''mBa / '''+str("{:.1f}".format(pressurepsi))+'''PSI  -  T: '''+str("{:.1f}".format(temperature))+'''c  -  B: '''+str(batterypercentage)+'''%
font_name=serif
layout_align=top,left
horizontal_margin=50
pt=20''')
        f.close()
        os.system("cat /home/pi/subtitle.txt > /home/pi/picam/hooks/subtitle")
    time.sleep(0.5)
```

iizukanao, 2021-03-18:
@Boeeerb Thanks for providing a script. Confirmed the issue and looking into it.

Boeeerb, 2021-04-28:
@iizukanao Did you find anything I can try to reduce?

I noticed the timestamp doesn't have any flicker in my testing, and peeking through the source they seem to draw using the same function

Boeeerb, 2021-04-29:
I've had it running for around 10 hours today, and skimming through the footage, and happy to report I didn't see any flickering, so happy to say it's been resolved.

Many thanks for putting this fantastic software together

iizukanao, 2021-04-30:
@Boeeerb Glad to hear that!

closed 2021-04-30 by iizukanao