iizukanao/picam #162: Subtitles

planet4, 2021-05-25:
I am about to give up. When running Picam manually subtitles works fine but after using init.d and a restart it does not work at all. I have tried all kind of tricks just to add a title next to the time. What have I missed? Must Picam use system.d instead? What is the proper way of autostarting picam?

iizukanao, 2021-05-25:
@planet4 Do you do `cd` to picam directory before you execute `picam` command? Also, if possible, please share your script here.

alemelis, 2021-10-02:
@planet4 `sleep` did the trick for me

this is my script to run picam on my raspberrypi

```
/home/pi/picam/picam -o /run/shm/hls \
	--vfr \
	--autoex \
	--time \
	--alsadev hw:1,0 &
sleep 3

touch /home/pi/hooks/subtitle
sleep 10

cat /home/pi/subs > /home/pi/hooks/subtitle
```

hope it helps!

(possibly related to #158 ?)