iizukanao/picam #164: Switch between two cameras

98Matthias, 2021-07-13:
Hi,

the Compute Module IO Board has two CSI-2 camera interfaces.
The supplied camera applications raspivid and raspistill have the -cs (--camselect) option to specify which camera should be used.

Is it possible to add this parameter?

Greetings,
Matthias

iizukanao, 2022-07-07:
Sorry for the late reply. Yes, it is possible to add the parameter for camera selection. If you are still interested, please let me know.

98Matthias, 2022-07-07:
Yes, that would be great! I have read that the option changed to --camera in libcamera.

[image]

iizukanao, 2022-07-07:
Thanks for your comment!

iizukanao, 2022-07-10:
@98Matthias I have built a picam binary that supports `--camera` option. Can you check if it works?
[picam-2.0.1-armv7l.tar.gz](https://github.com/iizukanao/picam/files/9078385/picam-2.0.1-armv7l.tar.gz)

Usage:
```
--camera <num>      Choose the camera to use. Use --query to list the cameras.
```

98Matthias, 2022-07-10:
Thanks a lot! I will be able to test it in a few days 👍

98Matthias, 2022-07-15:
It works perfectly. Thanks for your effort!

iizukanao, 2022-07-15:
@98Matthias Thank you for your feedback! I will release it as version 2.0.1 soon.

closed 2022-07-15 by iizukanao