iizukanao/picam #166: New Raspberry Pi OS "Bullseye" and picam

gamer1966, 2021-11-14:
Hi,

under the new Raspberry Pi OS "Bullseye" picam doesn't work.
The driver used to access camera modules has been replaced.

So, at this moment, do not upgrade to "Bullseye" if you want to use picam.

@iizukanao: is there a chance to implement the new changes to your software?

CharlesBinard, 2021-11-19:
@gamer1966  have you found an alternative or a fix?

gamer1966, 2021-11-19:
> @gamer1966 have you found an alternative or a fix?

Sorry, I'm not the developer.

As you can read [here](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps),  the new software library 'libcamera' drive the camera system.
A brand new version of picam may be required.

mistes32, 2022-01-24:
@gamer1966
You should enable legacy camera support in raspi-config interfaces.

gamer1966, 2022-01-28:
@mistes32
Thank you for the hint.

Before it is possible to enable legacy camera you have to update raspi-config (**8 Update   Update this tool to the latest version**)

But that is not enough. There are missing libs and files in /opt/vc. 
The following set of commands help.

_git clone https://github.com/raspberrypi/firmware
sudo cp -r firmware/hardfp/opt/vc /opt/
echo "/opt/vc/lib" > opt-vc.conf
sudo chown root.root opt-vc.conf
sudo mv opt-vc.conf /etc/ld.so.conf.d/
sudo ldconfig_

(thanks to peter64 @ https://forums.raspberrypi.com/viewtopic.php?t=317888&start=25)

iizukanao, 2022-07-07:
Sorry for the late response. picam version 2.0.0 for Bullseye OS (with libcamera) is now available. Legacy camera stack is no longer used.
https://github.com/iizukanao/picam/releases/tag/v2.0.0

closed 2022-07-07 by iizukanao