iizukanao/picam #167: error while loading shared libraries: libopenmaxil.so

DrFailbucket, 2022-01-28:
Could someone please help me? 

When i try "./picam --alsadev hw:1,0" then this error shows up:
"./picam: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory"

jlar0che, 2022-02-15:
Same issue here. Seems like picam doesn't work with the newest version of Raspbian (Bullseye).

bxmatus, 2022-04-01:
This file works for me: https://github.com/raspberrypi/rpi-firmware/blob/74b4d8722026cf1db5f8a3953bca2639dc548667/vc/hardfp/opt/vc/lib/libopenmaxil.so

simonborgithub, 2022-04-04:
This worked for me: sudo ln -s /opt/vc/lib/libopenmaxil.so /usr/lib/libopenmaxil.so 

(from: https://github.com/aws-samples/amazon-kinesis-video-streams-demos/blob/master/producer-cpp/docker-raspberry-pi/Dockerfile)

iizukanao, 2022-06-21:
Sorry for the inconvenience. I finally got some time and I'm reworking picam to use libcamera instead of the legacy camera/encoder libraries (OpenMAX and MMAL). Please wait a little longer.

iizukanao, 2022-07-07:
picam version 2.0.0 has been released with libcamera support. libopenmaxil.so and legacy camera stack are no longer needed.
https://github.com/iizukanao/picam/releases/tag/v2.0.0

closed 2022-07-07 by iizukanao