iizukanao/picam #169: Lagging in video and microphone error: buffer overrun

furkanbalikci, 2022-06-18:
Hello, 
I am trying to record audio and video with picam. I am using Rpi Zero as device. I am using picamera for video and external mems microphone for audio. There are lags in the test videos that I am moving. And I keep getting the "microphone error: buffer overrun" error. I'm not sure if these two are related.
What I really need is to be able to record video and audio synchronously.
Do you have any suggestions for these topics?
Thanks.

iizukanao, 2022-06-21:
Hello @furkanbalikci, thanks for your feedback.
It seems that the data transfer rate from the microphone is lower than it should be. Please use the `top` or `glances` commands to make sure the Pi is not doing too much work while recording (check CPU, SWAP, and disk I/O).

Btw, how many audio channels can your microphone record?

furkanbalikci, 2022-06-21:
Hi @iizukanao , thanks for your quick answer. I bough a new voice card, its using same mic. For now I don't get microphone error but lag problem is still going on. 
Also I checked  CPU, its around %50, SWAP is %0, and for disk I/O i used iostat -d 2 command output is bellow. After new voice card I use alsa command with hw:0 not hw:0,0. its look better but I think main problem is frame dropping, what can I do for this problem. Its like urgent situation, I am waiting your reply. 

Thanks for everything.

iizukanao, 2022-06-21:
@furkanbalikci Hmm, I have no idea why that happens. How many seconds of lag are there? Are you experiencing both lag (video and audio are out of sync) and frame dropping (intermittent sound) at the same time? Can you record audio with `arecord` command without problem?

furkanbalikci, 2022-06-21:
@iizukanao thanks for quick answer. Now I tested at outside. I walked with like 10 minutes. Now I have 2 situation. 
1) The quality of video looks bad. I am using default (1280/720 alsa fps is defauld)  but its look like 480p. But  when I not moving much better. 
2) Still sometimes frame is dropping. I tried fps with 15,25 and it did not effect the frame dropping. 
What could be causing this? Could the sound have an effect on this?

furkanbalikci, 2022-06-21:
@iizukanao I solved the video resolution drop problem by increasing the bitrate. But video stuttering is still present (I defined this event as framedropping earlier). Please help with this :(

iizukanao, 2022-06-22:
@furkanbalikci If you don't see frame drops in the preview (`-p` option), maybe the SD card write performance is too slow or some other program is overloading the disk I/O and not recording the video properly. If you still see `microphone error` message, there may be a problem with your ALSA configuration or hardware.

furkanbalikci, 2022-06-23:
Hello again @iizukanao, this is my full verbose output. As you can see sometime is dropping. Does this mean anything to you? The camera stays in a fixed position meanwhile.

pi@raspberrypi:~/picam $ ./picam --alsadev hw:0 --videobitrate 5000000 --time --verbose
video_width=1280
video_height=720
video_fps=30.000000
fr_q16=1966080
video_pts_step=3000
video_gop_size=30
video_rotation=0
video_hflip=0
video_vflip=0
video_bitrate=5000000
video_avc_profile=constrained_baseline
video_avc_level=3.1
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
alsa_dev=hw:0
audio_channels=1
audio_sample_rate=48000
audio_bitrate=40000
audio_volume_multiply=1.000000
is_hlsout_enabled=0
is_hls_encryption_enabled=0
hls_keyframes_per_segment=1
hls_number_of_segments=3
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
is_timestamp_enabled=1
timestamp_format=%a %b %d %l:%M:%S %p
timestamp_layout=15
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
timestamp_pos_x=0
timestamp_pos_y=0
timestamp_text_align=1
[... 244 lines cut ...]
.............................. 30.10 fps k=149 a-v=3030 c-a=717923 u=10 d=0 pts=14448810
.............................. 30.10 fps k=150 a-v=3270 c-a=717623 u=10 d=0 pts=14538810
.............................. 30.20 fps k=151 a-v=3510 c-a=717434 u=10 d=0 pts=14632320
......F........................ 29.11 fps k=152 a-v=5670 c-a=718288 u=10 d=0 pts=14718810
.............................. 30.09 fps k=153 a-v=5910 c-a=718017 u=10 d=0 pts=14808810
.............................. 30.16 fps k=154 a-v=6150 c-a=717630 u=10 d=0 pts=14904960
.............................. 30.16 fps k=155 a-v=6390 c-a=717539 u=10 d=0 pts=14995200
.............................. 30.26 fps k=156 a-v=4710 c-a=718737 u=10 d=0 pts=15078810
.............................. 30.05 fps k=157 a-v=4950 c-a=718659 u=10 d=0 pts=15168810
.....F.vR104070........................ 14.55 fps k=158 a-v=-2880 c-a=718269 u=10 d=0 pts=15362880
.............................. 30.09 fps k=159 a-v=-2640 c-a=718025 u=10 d=0 pts=15452880
.............................. 30.12 fps k=160 a-v=-2400 c-a=717680 u=10 d=0 pts=15542880
.............................. 29.90 fps k=161 a-v=-2160 c-a=718003 u=10 d=0 pts=15630720
.............................. 30.39 fps k=162 a-v=-3840 c-a=718793 u=10 d=0 pts=15722880
.............................. 29.92 fps k=163 a-v=-1680 c-a=717569 u=10 d=0 pts=15811200
...F.vR95760.......................... 14.60 fps k=164 a-v=-3120 c-a=718438 u=10 d=0 pts=15998640
.............................. 30.10 fps k=165 a-v=-2880 c-a=718188 u=10 d=0 pts=16088640
.............................. 30.11 fps k=166 a-v=-2640 c-a=717886 u=10 d=0 pts=16178640
.............................. 30.14 fps k=167 a-v=-2400 c-a=717518 u=10 d=0 pts=16268640
.............................. 30.18 fps k=168 a-v=-4080 c-a=718918 u=10 d=0 pts=16358640
.............................. 29.98 fps k=169 a-v=-1920 c-a=717574 u=10 d=0 pts=16446720
.F............................. 29.39 fps k=170 a-v=240 c-a=718156 u=10 d=0 pts=16538640
.............................. 30.19 fps k=171 a-v=480 c-a=717568 u=10 d=0 pts=16628640
.............................. 29.99 fps k=172 a-v=720 c-a=717596 u=10 d=0 pts=16718640
.............................. 30.25 fps k=173 a-v=-960 c-a=718794 u=10 d=0 pts=16808640

closed 2022-06-27 by furkanbalikci