iizukanao/picam #171: Runnig on /run/shm

furkanbalikci, 2022-06-27:
@iizukanao my fps problem still going on. When I read your usage guide again I saw your suggestion "You can take advantage of RAM drive `(/run/shm/)` and reduce access to SD card. It also provides better quality of recording."  When I googled the folder name, I saw that this path is now used as `/dev/shm/`. You can check it too if you want. Anyway, what I did was to write `/dev/shm/` instead of `/run/shm/ `in the `make_dirs.sh` file. Then I copied the `picam-1.4.11-binary/picam` file into `/run/shm/` and I think I can use the advantage of ram. Am I missing something in these steps? Or do you have any other suggestions to use it more effectively? If it helps, I can share the stutters with the fps drops with you.

iizukanao, 2022-06-28:
Hello @furkanbalikci, sorry for not replying to the previous issue. If **/run/shm** does not exist in your filesystem, you can just create a symbolic link with `sudo ln -s /dev/shm /run/shm` and all scripts should work fine.

furkanbalikci, 2022-06-28:
I am not using destkop version. 

Here,  output of` ls -lR /dev/shm/`

```
/dev/shm/:
total 0
lrwxrwxrwx 1 root root 9 Jun 28 12:11 shm -> /dev/shm/
```

I am using Pi with ssh. 
_sudo raspi-config > System Options > Boot > Console, then reboot._  
After this process, I am not sure that I could still connect via ssh. Even though I'm not using the desktop version, Should I still do it.

iizukanao, 2022-06-28:
If you can confirm that the GUI (graphical login screen) is not running, you don't have to reboot it.

furkanbalikci, 2022-06-28:
> @furkanbalikci That folder structure is incorrect and RAM disk is not used. Please run the following commands to correct the `/run/shm`.
> 
> ```shell
> sudo rm -rf /dev/shm/*
> sudo rm -rf /run/shm
> sudo ln -s /dev/shm /run/shm
> ```
> 
> Do not append `/` at the end of the above commands.

I tried your recommended steps but now both folders are empty.

iizukanao, 2022-06-28:
@furkanbalikci Run `./make_dirs.sh` then run `picam`

furkanbalikci, 2022-06-28:
After this my full output is bellow. It's not a smooth video. FPS dropping I can catch easily. 

`..................disk_usage=3% .start rec to rec/archive/2022-06-28_14-13-54.ts
........... 30.20 fps k=26 a-v=4470 c-a=1123 u=1 d=0 pts=2349450
......caught up........................ 30.10 fps k=27 a-v=4710 c-a=855 u=1 d=0 pts=2439450
.............................. 30.13 fps k=28 a-v=4950 c-a=580 u=1 d=0 pts=2529450
.............................. 30.10 fps k=29 a-v=5190 c-a=389 u=1 d=0 pts=2619450
.............................. 30.16 fps k=30 a-v=5430 c-a=48 u=1 d=0 pts=2714880
............................F.. 29.02 fps k=31 a-v=7590 c-a=1201 u=1 d=0 pts=2799450
.............................. 30.49 fps k=32 a-v=7830 c-a=35 u=1 d=0 pts=2897280
.............................. 30.31 fps k=33 a-v=6150 c-a=1077 u=1 d=0 pts=2979450
.............................. 30.15 fps k=34 a-v=6390 c-a=1088 u=1 d=0 pts=3069450
.............................. 30.02 fps k=35 a-v=6630 c-a=1053 u=1 d=0 pts=3159450
.............................. 30.09 fps k=36 a-v=6870 c-a=811 u=1 d=0 pts=3249450
...........................F.vSPEED_UP(16950).. 27.56 fps k=37 a-v=14340 c-a=1216 u=2 d=0 pts=3339900
.............................. 30.28 fps k=38 a-v=10080 c-a=863 u=2 d=0 pts=3434400
.............................. 30.00 fps k=39 a-v=5820 c-a=983 u=2 d=0 pts=3528900
......................vNORMAL........ 30.13 fps k=40 a-v=2910 c-a=677 u=2 d=0 pts=3622050
.............................. 30.13 fps k=41 a-v=3150 c-a=382 u=2 d=0 pts=3712050
.............................. 30.15 fps k=42 a-v=3390 c-a=-48 u=2 d=0 pts=3802050
.........................FvSPEED_UP(12870)..... 28.20 fps k=43 a-v=8490 c-a=-9 u=3 d=0 pts=3892950
.............................. 30.23 fps k=44 a-v=2310 c-a=1227 u=3 d=0 pts=3987450
.......vNORMAL....................... 30.23 fps k=45 a-v=1650 c-a=1032 u=3 d=0 pts=4078350
.............................. 30.07 fps k=46 a-v=1890 c-a=886 u=3 d=0 pts=4168350
.............................. 30.08 fps k=47 a-v=2130 c-a=703 u=3 d=0 pts=4258350
.............................. 30.10 fps k=48 a-v=2370 c-a=535 u=3 d=0 pts=4348350
.......................FvSPEED_UP(10170)....... 28.23 fps k=49 a-v=7170 c-a=502 u=4 d=0 pts=4439550
............................vNORMAL.. 30.11 fps k=50 a-v=3360 c-a=215 u=4 d=0 pts=4533600
.............................. 30.26 fps k=51 a-v=3600 c-a=-164 u=4 d=0 pts=4627200
.............................. 30.21 fps k=52 a-v=1920 c-a=1181 u=4 d=0 pts=4713600
.............................. 30.23 fps k=53 a-v=2160 c-a=940 u=4 d=0 pts=4803600
.............................. 30.01 fps k=54 a-v=2400 c-a=973 u=4 d=0 pts=4893600
.....................FvSPEED_UP(10440)......... 28.21 fps k=55 a-v=6900 c-a=1030 u=5 d=0 pts=4985100
.............................. 30.12 fps k=56 a-v=2640 c-a=682 u=5 d=0 pts=5079600
.vNORMAL............................. 30.09 fps k=57 a-v=2880 c-a=459 u=5 d=0 pts=5169600
.............................. 30.11 fps k=58 a-v=3120 c-a=172 u=5 d=0 pts=5259600
.............................. 30.27 fps k=59 a-v=3360 c-a=-165 u=5 d=0 pts=5352960
.............................. 30.18 fps k=60 a-v=1680 c-a=1225 u=5 d=0 pts=5439600
...................FvSPEED_UP(9960)........... 28.39 fps k=61 a-v=5880 c-a=1019 u=6 d=0 pts=5531400
.............................. 22.49 fps k=62 a-v=32340 c-a=404 u=6 d=0 pts=5625900
.............................. 30.13 fps k=63 a-v=28080 c-a=56 u=6 d=0 pts=5720400
.............................. 30.06 fps k=64 a-v=23820 c-a=-92 u=6 d=0 pts=5814900
.............................. 30.22 fps k=65 a-v=17640 c-a=1207 u=6 d=0 pts=5909400
.............................. 30.24 fps k=66 a-v=13380 c-a=961 u=6 d=0 pts=6003900
.......F....................... 28.23 fps k=67 a-v=14880 c-a=910 u=6 d=0 pts=6098400
.............................. 30.07 fps k=68 a-v=10620 c-a=798 u=6 d=0 pts=6192900
.............................. 30.06 fps k=69 a-v=6360 c-a=673 u=6 d=0 pts=6287400
........................vNORMAL...... 30.18 fps k=70 a-v=3150 c-a=250 u=6 d=0 pts=6380850
.............................. 30.12 fps k=71 a-v=3390 c-a=-77 u=6 d=0 pts=6470850
.............................. 30.25 fps k=72 a-v=1710 c-a=1108 u=6 d=0 pts=6560850
.....F.vSPEED_UP(12510)........................ 28.30 fps k=73 a-v=3960 c-a=1255 u=7 d=0 pts=6654600
................vNORMAL.............. 30.26 fps k=74 a-v=1950 c-a=940 u=7 d=0 pts=6746850
.............................. 30.08 fps k=75 a-v=2190 c-a=779 u=7 d=0 pts=6836850
.............................. 30.07 fps k=76 a-v=2430 c-a=662 u=7 d=0 pts=6926850
.............................. 30.11 fps k=77 a-v=2670 c-a=385 u=7 d=0 pts=7016850
..........stop rec
copy symlink(archive/2022-06-28_14-13-54.ts, rec/2022-06-28_14-13-54.ts)
unlink.vSPEED_UP(9990)................... 28.23 fps k=78 a-v=5670 c-a=288 u=8 d=0 pts=7109850
`

maybe I should reinstall everything then additionally use just `sudo ln -s /dev/shm /run/shm` command ? 

What do you think about this topic:

> When I examine the speedup parts here in the source code, I see that it is related to PTS. Does this mean that it is causing fps drops as it tries to match the audio with the video? Is there anything I can do to improve this?

iizukanao, 2022-06-28:
> When I examine the speedup parts here in the source code, I see that it is related to PTS. Does this mean that it is causing fps drops as it tries to match the audio with the video? Is there anything I can do to improve this?

`vSPEED_UP` appears when the video frame rate is lower than expected (less than 30 fps in your case), and picam is internally skipping some video PTS for the missing video frames to keep the audio and video PTS in sync.

From the picam output log, I think the write speed of the SD card is too slow. Please use a faster SD card.

furkanbalikci, 2022-06-28:
I am using now **samsung evo plus micro sd card 256gb**. What would you suggest instead of this.

iizukanao, 2022-06-28:
I'm using a Toshiba Exceria series. I can't guarantee that it will work in your case.
Btw, what is the output of `sudo dd if=/dev/zero of=/home/pi/test bs=8k count=50k conv=fsync; sudo rm -f /home/pi/test` ?

My result is
```
51200+0 records in
51200+0 records out
419430400 bytes (419 MB, 400 MiB) copied, 19.5284 s, 21.5 MB/s
```

furkanbalikci, 2022-06-28:
`./picam --alsadev hw:0 --time --videobitrate 5000000 --verbose`

furkanbalikci, 2022-06-28:
Can you send to me link of your SD card?

iizukanao, 2022-06-29:
@furkanbalikci https://www.amazon.co.jp/%E6%9D%B1%E8%8A%9D-Toshiba-microSDHC-UHS-I-EXCERIA%E8%B6%85%E9%AB%98%E9%80%9F95MB/dp/B00C67VPGI

furkanbalikci, 2022-06-29:
@iizukanao  what rpi model are you using?

iizukanao, 2022-06-29:
@furkanbalikci I'm using pi 4B (4GB) + camera module v2 + microphone listed [here](https://github.com/iizukanao/picam#usb-microphone).
command line is `./picam --alsadev hw:1,0`