iizukanao/picam #172: HLS does not appear to be functioning

jmarshall120, 2022-07-18:
With the release of picam 2.0, I have migrated to Bullseye and completed the upgrade to the new picam version. Unfortunately, I am having issues using HLS.

1. I have included the required section in my _/etc/nginx/sites-available/default_ file:
`server {
...
       location /hls/ {
		root /run/shm;
       }
...
}`

3. I have temporarily disabled the picam init.d script and I am executing the following (in the picam folder), with it's output all looking fine (_make_dirs.sh_ is executed prior):
`sudo ./picam -o /run/shm/hls --alsadev hw:1,0
created directory: ./rec/tmp
[0:12:08.177808082] [675]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:12:08.280450654] [677]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:12:08.282136701] [675]  INFO Camera camera.cpp:1029 configuring streams: (0) 1920x1080-YUV420
[0:12:08.282851078] [677]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
capturing started`

4. I then attempt to access the stream using Firefox from a client computer and am met with the nginx "403 Forbidden" page
5. I then attempt to stream to VLC on the client using `http://192.168.2.8/hls/index.m3u8` which is throwing a "VLC is unable to open..." error.

Here is the output of _/var/log/nginx/access.log_
`192.168.2.166 - - [18/Jul/2022:20:17:52 +1000] "GET /hls/ HTTP/1.1" 403 125 "-" "Mozilla/5.0 (X11; Linux x86_64; rv:102.0) Gecko/20100101 Firefox/102.0"
192.168.2.166 - - [18/Jul/2022:20:20:29 +1000] "GET /hls/index.m3u8 HTTP/1.1" 404 153 "-" "VLC/3.0.17.2 LibVLC/3.0.17.2"`

Here is the output of _/var/log/nginx/error.log_
`2022/07/18 20:17:52 [error] 460#460: *1 directory index of "/run/shm/hls/" is forbidden, client: 192.168.2.166, server: _, request: "GET /hls/ HTTP/1.1", host: "192.168.2.8"
2022/07/18 20:20:29 [error] 460#460: *2 open() "/run/shm/hls/index.m3u8" failed (2: No such file or directory), client: 192.168.2.166, server: _, request: "GET /hls/index.m3u8 HTTP/1.1", host: "192.168.2.8"`

I have also used `chown -R www-data:www-data` and `chmod 744` on the _/run/shm/hls directory_

My guess is that the index.m3u8 stream file isn't being created in the _/run/shm/hls_ folder. Any ideas?

jmarshall120, 2022-07-20:
Sorry, I knew I forgot to post some info the other day.  /run/shm/hls is empty while picam is running:
`pi@rpiz2:~ $ ls /run/shm/hls/ -al`
`total 0`
`drwxr-xr-x 2 www-data www-data  40 Jul 20 19:05   .`
`drwxrwxrwt 6 root     root     120 Jul 20 19:05   ..`

I've tried `sudo ./picam -o /run/shm/hls --alsadev hw:1,0` with and without `sudo`. The directory is empty both ways.

There's no configuration in nginx for the 403 error. My intuition is telling me that picam is not creating the output in the /run/shm/hls directory for some reason, despite the output of the picam command not showing any errors.

////////////////////////////////
**I've narrowed the problem down to likely audio-related**

So I've managed to get the stream partially working; however, I had to disable audio capture. I'm using this command now, which I am able to stream to a client: `sudo ./picam -o /run/shm/hls --time`

This also results in the index.m3u8 and multiple .ts files being in /run/shm/hls.

When specifying the audio device there have been no errors thrown. I've also confirmed my user is a member of the audio group:
`pi@rpiz2:~ $ groups`
`pi adm dialout cdrom sudo audio video plugdev games users input render netdev gpio i2c spi`

iizukanao, 2022-07-20:
Thanks for your feedback. Unfortunately, I am unable to reproduce the issue regardless of the availability of audio. HLS files are successfully created with or without audio. Can you post the full picam output with the option `--verbose`?

jmarshall120, 2022-07-23:
I still haven't worked this out. The /run/shm/hls directory only fills up when --noaudio is used (or the wrong asladev is sepcified so it disables audio automatically)

`pi@tpiz2w:~$ cat run_picam.sh
sudo bash /home/pi/make_dirs.sh
cd /home/pi/picam && sudo ./picam -o /run/shm/hls --time --alsadev hw:1,0 --verbose > /var/log/picam.log 2>&1
pi@rpiz2:~ $ ./run_picam.sh 
^C^C
pi@rpiz2:~$ cat /var/log/picam.log
video_width=1920
video_height=1080
video_fps=30.000000
video_pts_step=3000
video_gop_size=30
video_hflip=0
video_vflip=0
video_bitrate=4500000
video_avc_profile=baseline
video_avc_level=4.1
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
alsa_dev=hw:1,0
audio_channels=1
audio_sample_rate=48000
audio_bitrate=40000
audio_volume_multiply=5.000000
is_hlsout_enabled=1
is_hls_encryption_enabled=0
hls_keyframes_per_segment=1
hls_number_of_segments=3
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/hls
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=�)N�
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
is_timestamp_enabled=1
timestamp_format=%a %b %d %l:%M:%S %p
timestamp_layout=15
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
[... 138 lines cut ...]
.............................. 30.35 fps k=18 a-v=-3360 c-a=-29195 u=0 d=0 pts=1650720
.............................. 29.75 fps k=19 a-v=-1200 c-a=-30590 u=0 d=0 pts=1740720
.............................. 29.99 fps k=20 a-v=-2880 c-a=-28871 u=0 d=0 pts=1830720
...............stop requested (signal=2)
.Halting
StopCamera
Camera stopped!
VideoEncoder closed
stopAudioThread begin
joining audio thread
joined audio thread
teardown_audio_capture_device
stopAudioThread end
stopRecThread begin
stopRecThread end
stop_watching_hooks
pthread_join hooks_thread
shutdown successful
Closing Libcamera application (preview frames displayed 617, dropped 0
StopCamera
Camera stopped!
preview: stopPreview begin
preview: stopPreview end
Tearing down requests, buffers and configuration
Camera closed`

iizukanao, 2022-07-23:
@jmarshall120 Are you able to record audio from the microphone with the `arecord` command?

jmarshall120, 2022-07-23:
Yep, able to get an audio recording using
`arecord --format=S16_LE --rate=48100 --file-type=wav out.wav`

I've tried with 16000 too which also works (quality varies of course).

Thanks so much for your continued help!

iizukanao, 2022-07-23:
Regarding `arecord` command, `--rate=48000` work as well?

jmarshall120, 2022-07-23:
Yep, all bit rates appear to work, just the quality is bad but I believe that's just the USB microphone.

Maybe time for a fresh Raspian install and go from scratch again?

iizukanao, 2022-07-23:
@jmarshall120 Hmm, I don't know if starting over from scratch will solve the problem. Does it record audio when recorded with the following command while picam is running?
```sh
touch hooks/start_record && sleep 10 && touch hooks/stop_record
```
Recorded files will be created in the rec/archive/ directory.

jmarshall120, 2022-07-31:
@iizukanao Sorry for my slow response. Yes, that command does create a file that I can play back.

iizukanao, 2022-08-01:
@jmarshall120 Hmm, I have no idea why only HLS does not work. If you could provide the SD card image, I might be able to look into it.

iizukanao, 2022-08-10:
My apologies. I have confirmed that this issue occurs on 64-bit OS. I will be fixing this issue soon.

iizukanao, 2022-08-10:
This bug has been fixed in version 2.0.4.

jmarshall120, 2022-08-11:
Thank you for all your hard work!

iizukanao, 2022-08-12:
Thank you for letting me know the issue!

commit c240b0ec (2022-08-10): Fix bug where is_hlsout_enabled does not work

Fixes #172