iizukanao/picam #174: Cannot end picam

bostiko, 2022-07-24:
Hi,
I run picam on several RPI. Now I want to upgrade to new version. I start again from scratch, install picam, copy my scripts. But now I have a small problem with picam. I start it automatically with a bash script:

`if [ "$1" != "" ]; then
    url="$1"
fi

ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a aac -f flv $url 2>> $log_ffmpeg &
sleep 1
/home/pi/picam/picam -alsadev plughw:1,0 --volume 2 --verbose --tcpout tcp://127.0.0.1:8181 >> $log_picam`

When I end the script, picam won't stop. Can you help me please?
Thank you :)

iizukanao, 2022-07-24:
@bostiko Thank you for your input. I think the problem is due to mishandling of SIGTERM. I have fixed the problem in this attached picam binary. Could you please check if it works as you intended?
[picam-2.0.3-armv7l.tar.gz](https://github.com/iizukanao/picam/files/9175931/picam-2.0.3-armv7l.tar.gz)

bostiko, 2022-07-25:
Hi @iizukanao, thank you for your quick reply. I am unable to test your binary, because my RPi runs on arch64 architecture :( Can you rebuild it for achr64 architecture please? Thank you :)

iizukanao, 2022-07-25:
@bostiko Sure! Please use this instead.
[picam-2.0.3-aarch64.tar.gz](https://github.com/iizukanao/picam/files/9178590/picam-2.0.3-aarch64.tar.gz)

bostiko, 2022-07-25:
Hallo, It now works. Thank you :)

closed 2022-07-25 by bostiko

iizukanao, 2022-07-25:
Great!