iizukanao/picam #175: After running for a few days, the process gets killed for OOM

BramGruneir, 2022-08-06:
Hi! 

Here's the command line from the service I setup:
`ExecStart=/home/me/picam/picam --tcpout tcp://127.0.0.1:8181 --time --avcprofile high --autoex --alsadev hw:2,0`

```Aug  4 01:22:47 babypi kernel: [298876.908936] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=picam,pid=5655,uid=0
Aug  4 01:22:47 babypi kernel: [298876.909013] Out of memory: Killed process 5655 (picam) total-vm:780700kB, anon-rss:617416kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:716kB oom_score_adj:0
Aug  4 01:22:47 babypi kernel: [298877.072194] oom_reaper: reaped process 5655 (picam), now anon-rss:0kB, file-rss:220kB, shmem-rss:0kB
Aug  5 20:28:32 babypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Aug  5 20:28:32 babypi kernel: [    0.000000] Linux version 5.15.32-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538
 SMP Thu Mar 31 19:38:48 BST 2022
```

This is running version 2.0.1.  I'm going to try to update to 2.0.3 and see if that helps but it didn't look the changes 

Other potentially useful info:
```
Raspberry Pi 3 Model B Plus Rev 1.3
Raspberry Pi reference 2022-04-04
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
5.15.32-v7+
```

iizukanao, 2022-08-06:
Hi, thanks for the info. It looks like there is a memory leak somewhere in picam. I will look into it.

iizukanao, 2022-08-10:
I have fixed a memory leak in version 2.0.4. Could you please check if the OOM no longer occurs?

BramGruneir, 2022-08-10:
Wonderful!  I'll give it a try tonight (and let you know in a few days if it fixed the problem).

BramGruneir, 2022-08-31:
All good!  I've had picam up and running for days at a time.  Thanks for quick turn around.

closed 2022-08-31 by BramGruneir

iizukanao, 2022-08-31:
Thanks for letting me know!