iizukanao/picam #176: Poor video quality when using Bullseye vs. Buster

feacluster, 2022-08-06:
I upgraded my pi4 to Bullseye and now my V2 camera is recording almost unwatchable videos. See this example:

https://youtu.be/jKHUdazLfBU?t=327

`~/picam/picam --alsadev hw:1,0 --fps 24 -v 1000000 -w 1640 -h 1232 `

compared to the same camera with Buster:

https://www.youtube.com/watch?v=r_DktnphecQ

`~/picam/picam --alsadev hw:1,0 --fps 24 -v 1000000 --mode 4 -w 1640 -h 1232`

Seems white balance or contrast issue?

iizukanao, 2022-08-08:
Could you please try adding `--avcprofile high` and see if it improves the image quality?

feacluster, 2022-08-08:
Thanks, that worked!

iizukanao, 2022-08-10:
Thank you for your confirmation. Starting with picam version 2.0.4 released today, `--avcprofile high` is the default.

closed 2022-08-10 by iizukanao