iizukanao/picam #178: Error: cannot set channel count for microphone (Invalid argument)

blogdemoi, 2022-08-12:
Hi,

I installed Picam on a Raspberry Pi 3. The camera is a Pi camera and the microphone is a usb "PlayStation Eye". When I run `./picam --alsadev hw:1,0` I have this error:

> picam ./picam --alsadev hw:1,0
> error: cannot set channel count for microphone (Invalid argument)

> picam arecord -l
> List of CAPTURE Hardware Devices
> card 1: CameraB409241 [USB Camera-B4.09.24.1], device 0: USB Audio [USB Audio]
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0

What can be wrong?

Thank you for any suggestions.

iizukanao, 2022-08-13:
Hi, please follow the guide below to create an ALSA virtual device that mixes 4 microphones into one, then use that virtual device (`cap`) with picam.
https://www.cnx-software.com/2019/08/30/using-sony-ps3-eye-camera-as-an-inexpensive-microphone-array/

blogdemoi, 2022-08-13:
Thank you. I followed the guide you mentioned. However now I have:

> picam ./picam --alsadev hw:1,cap
> error creating state dir (./state): File exists

iizukanao, 2022-08-14:
I think the `state` in your filesystem is a broken symbolic link. Fix the symbolic link or remove it with `rm state`.

lukaszgielecinski, 2022-10-24:
I have a 2xElectret condenser microphone and I added `/etc/asound.conf` file with the following content to fix the same problem:
```
pcm.channelsum {
 type plug
 slave {
  pcm "hw:1,0"
  channels 2
 }
 route_policy sum
}
```
I rebooted pi and I use it like that:
```
./picam --alsadev channelsum
```

blogdemoi, 2022-10-25:
> I have a 2xElectret condenser microphone and to fix the same problem I added /etc/asound.conf file with the following content

Thank you, I haven't had time till now to go further, I will try when I have more time.