iizukanao/picam #179: Sound error when stream to youtube

bostiko, 2022-08-13:
Hallo,
first of all, thanks a lot for your work, @iizukanao .
Currently after reinstall and update to last version 2.0.4 I encounter an strange issue. I am able to send FullHD stream to youtube, but the sound sounds very strange. I record this example for you:
https://www.youtube.com/watch?v=Ozr5QipRClw&ab_channel=Ond%C5%99ejBostik
(From 0:00:20 I said: Test, test, test, one, two, three, test, test, test).

The script I am using for start of the stream is this:
```
url="rtmp://a.rtmp.youtube.com/live2/"

log_ffmpeg="/var/log/ffmpeg.log"
log_picam="/var/log/picam.log"

if [ "$1" != "" ]; then
    url="$1"
fi

echo "$(date) @ $(hostname) @ $url" | tee -a $log_picam $log_ffmpeg

ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a copy -f flv $url 2>> $log_ffmpeg &
sleep 1
/home/pi/picam/picam --alsadev hw:1,0 -c 2 --volume 2 --verbose --hdmi 1 --preview --tcpout tcp://127.0.0.1:8181 >> $log_picam
```

When I try run picam only (eg. with ./picam --alsadev hw:1,0), the audio and video are ok.
Do you have any suggestion please?

iizukanao, 2022-08-14:
Hello, thanks for using the latest version. There seems to be a problem with the microphone settings, such as the number of channels or sample rate. Is it possible to use only one microphone channel (`-c 1`)? Also, did streaming to YouTube work before picam version 2?

bostiko, 2022-08-18:
Hallo, thanks to your suggestions. I changed to only one microphone channel as suggested by you and it now works great.
(After that I also update picam to the latest picam version 2.0.5 and it also works.)
Thank you :)

closed 2022-08-18 by bostiko

iizukanao, 2022-08-18:
Excellent! Thank you for letting me know.