iizukanao/picam #180: Failed to start

Regradert, 2022-09-07:
Hello. I just installed it and it gives me this error when trying to test it. I do not know the motive:

`./picam --alsadev hw:1,0 -w 1920 -h 1080 -f 30 &`

`@raspberrypi:~/picam $ ./picam: symbol lookup error: ./picam: undefined symbol: _ZN9libcamera10ColorSpace4JpegE`

I don't know if I installed it wrong and I don't know what to try. Please, I would need a little help. Thanks.

DeborahFehr, 2022-09-08:
I have the same problem. This could be related to the renaming of the color_space Jpeg to Sycc in the libcamera package:
[https://git.libcamera.org/libcamera/libcamera.git/commit/?id=a17f0eddc61486989a375e34dcdcbb15c3152a4a](See here)

iizukanao, 2022-09-08:
Thank you for pointing this out. I will fix it shortly.

iizukanao, 2022-09-09:
This issue has been fixed in version 2.0.6.

closed 2022-09-09 by iizukanao

Regradert, 2022-09-10:
@iizukanao I happen to have the problem again. I installed it yesterday afternoon (on raspbian) and it worked perfectly, but now I need to install it on Dietpi OS and the error returns:

dietpi@DietPi:~/picam$ ./picam --alsadev hw:1,0
./picam: symbol lookup error: ./picam: undefined symbol: _ZN9libcamera10ColorSpace4SyccE

iizukanao, 2022-09-10:
@Regradert That may be because Dietpi OS uses an older version of libcamera. Build the latest libcamera yourself or use picam 2.0.5.

Regradert, 2022-09-11:
Many thanks @iizukanao. It works perfect now.

By the way, congratulations for picam, it is together with v4l2rtspserver the two best systems to transmit live video in good quality. I have tried all the ones available for the Raspberry. Picam seems to be the best since it sends audio by acc, you can add text and it allows you to crop the image (roi) it's amazing and from now on, I will recommend picam to everyone.

I miss a function that v4l2rtspserver has and I don't know if it can be done with picam. Is it possible to adjust brightness, contrast, exposure, etc. with the transmission running? I need it to change the brightness and contrast settings at dusk (or dawn) using a shell command. With v4l2rtspserver you can via these commands (without turning off the camera):

v4l2-ctl --set-ctrl brightness=60
v4l2-ctl --set-ctrl contrast=20

Thank you very much and congratulations again for this wonder !!

iizukanao, 2022-09-11:
@Regradert Thank you so much for your comment. I am very happy to hear that!
I think the dynamic control of brightness and contrast would be a nice addition. I am considering implementing it.

In the meantime, the white balance and exposure can be dynamically controlled with hooks.
White balance: https://github.com/iizukanao/picam#white-balance
Exposure: https://github.com/iizukanao/picam#exposure-control

Regradert, 2022-09-19:
> iizukanao

The system you have created is wonderful. It would be great to be able to control brightness/contrast/sharpness/color etc. dynamically. Let's see if you dare one day. Thank you very much !!

iizukanao, 2022-09-26:
@Regradert The brightness/contrast/saturation/sharpness hooks have been implemented in version 2.0.7. Please see [README](https://github.com/iizukanao/picam#hooks) for the usage.

iizukanao, 2022-09-26:
I will lock the conversation as the original issue has already been resolved.

commit 3d7f94c4 (2022-09-09): Use libcamera::ColorSpace::Sycc

Fixes #180