iizukanao/picam #182: hooks don't work if you start picam from init.d

Regradert, 2022-09-28:
Hello. First of all, thanks to the creator for adding the functions to adjust brightness/contrast/saturation, etc. without stopping the process. It works as a thousand wonders.

But I have a problem, I need to start picam automatically with init.d. This works great, but trying to adjust brightness, contrast... any hooks doesn't work at all. If I stop the picam process and instead of it starting automatically when the system boots, I do it manually from the console, it works perfectly.

But I need it to start automatically to change the brightness/contrast at sunrise and sunset. But it doesn't work, maybe I have something wrong.

**picam (in init.d):**

```
#! /bin/sh
# /etc/init.d/picam

### BEGIN INIT INFO
# Provides:          picam
# Required-Start:    $ALL
# Required-Stop:     $ALL
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: picam
### END INIT INFO

case "$1" in
    start)
        echo "Starting picam"
		sudo /home/pi/_comands/pican_run.sh
    ;;
    stop)
        echo "Stopping picam"
        killall picam
        killall -9 picam
    ;;
    *)
        echo "Usage: /etc/init.d/picam {start|stop}"
        exit 1
    ;;
esac
exit 0
```

**pican_run.sh**

```
sudo /home/pi/_comandos/make_dirs.sh

sudo /home/pi/picam/picam -o /run/shm/hls \
--videobitrate 2000000 \
--f 25 \
--metering center \
--alsadev hw:1,0 \
--audiobitrate 20000 \
--channels 1 \
--volume 4.0 \
--samplerate 44100 \
--time \
--timeformat "%H:%M:%S %d-%m-%G" \
--timelayout right \
--timept 18.0 \
--brightness 0.5 \
--contrast 1.8 
```

Once the system is booted, if I access from the terminal and put:

sudo echo 1.0 > hooks/brightness

It ignores me, but if I stop the picam process and run it directly in the terminal to start it again, the hooks do work.

I hope I have explained myself well, I am using the Google translator. Thank you very much and long live Picam.

iizukanao, 2022-09-28:
You need to `cd` first, and then start picam.
So, change `sudo /home/pi/picam/picam` to `cd /home/pi/picam && sudo ./picam`

Regradert, 2022-09-28:
Yeah. Many thanks @iizukanao With how simple it was and I've been thinking about it all morning. Wonderful.

closed 2022-09-28 by Regradert

iizukanao, 2022-09-29:
@Regradert No problem! Thank you for continuing to use picam.