iizukanao/picam #183: Audio sounds in slow motion

Regradert, 2022-10-11:
Hello. I have recently purchased another audio card for my Raspberry Pi, a [Sound Blaster Play! 3](https://es.creative.com/p/audio-enthusiasts/sound-blaster-play-3) but the audio sounds like it's playing in slow motion. I have tried changing the samplerate and there is no difference, I have also adjusted different bitrates and nothing.

I have two cards, with the previous one it sounds perfectly, but with this one it sounds like this and it's very strange. I have made a test audio recording outside of Picam:

`arecord -f S16_LE -c 2 -d 10 -r 44100 --device="hw:1,0" /home/pi/test.wav`

And it sounds perfect, no slow motion sound, but with Picam I have this problem. What could be happening?

iizukanao, 2022-10-12:
Does your sound card's microphone have any special specifications compared to a regular USB microphone?
One alternative is to create a virtual ALSA device instead of use `hw:1,0` directly.

Regradert, 2022-10-13:
Hi mate @iizukanao The microphone is a normal minijack, with the other card I have it works perfectly. The other card is this one, it's a very basic one but it works fine, I also have a micro usb and it works fine too. It is rare that Creative has this problem:

[https://www.mytrendyphone.es/images2/USB-Sound-Card-14052015-01-p.jpg](url)

I am going to try to create a virtual device as you say and I will tell you about it. Thank you very much.

brainrecall, 2022-12-15:
I had the same issue, and a virtual device fixed it.

For those who are looking how to do this, create a file at `~/.asoundrc`

Then add the following:
```
pcm.blaster {
  type plug
  slave {
    pcm "hw:2,0"
    channels 2
  }
  route_policy sum
}
```
Adjust the `hw:2,0` line to match your sound card number.

Then reboot, and use `--alsadev blaster` for picam.

tmqian, 2024-03-15:
Hello! Thanks so much for developing and supporting this awesome app!
I am also having the same issue with slow audio.

I tested the mic with 
`arecord -f S16_LE -c 2 -d 10 -r 44100 --device="hw:1,0" /home/pi/test.wav`
as above, and the file sounds normal.

So it is something about picam.
I am using HLS.

The above virtual device fix doesn't work for me.
I find that my .asoundrc gets deleted during/after reboot. The next time I try to run with `--alasdev blaster` there is no audio at all.

This is my picam command
`sudo /home/pi/picam/picam -o /run/shm/hls --samplerate 44100 --channels 2 –audiobitrate 96000 --time --vfr --autoex --alsadev blaster > ~/picam.log 2>&1`