iizukanao/picam #186: Unable to locate package libharfbuzz

leifnagtegaal, 2022-10-19:
Hi,
i try to install PiCam, but during the installation of the dependencies, i face into the issue, that apt is not able to locate the package libharfbuzz.
Has anyone a idea, how can i install it?

I using aRPi3b+ with bullsey x86

iizukanao, 2022-10-19:
Thanks for the feedback. Please run `sudo apt install libharfbuzz0b`

leifnagtegaal, 2022-10-19:
Thanks for your Help,
but, that brings me only one step ahead...

Next issue is: "./picam: error while loading shared libraries: libcamera.so.0: cannot open shared object file: No such file or directory"
i search for the lib, but i am a bit confused, because, on this page "https://www.raspberrypi.com/documentation/accessories/camera.html#getting-started" they mention, that When running a Raspberry Pi OS based on Bullseye, the 5 basic libcamera-apps are already installed.
But it seems, that is wrong?

Just a note, installing the other depended package "libfontconfig" returns: "Note, selecting 'libfontconfig1' instead of 'libfontconfig'" but i assume, thats not really an issue, correct?
Greeting

iizukanao, 2022-10-20:
Please see if the following commands will solve the problem.

```
sudo apt install libcamera0
sudo ldconfig
sudo reboot
```

> Just a note, installing the other depended package "libfontconfig" returns: "Note, selecting 'libfontconfig1' instead of 'libfontconfig'" but i assume, thats not really an issue, correct?

That's fine because `libfontconfig` is a virtual package.

leifnagtegaal, 2022-10-20:
Hi Nao,

LibCamera0 is already installed:
`pi@BabyPi:~ $ sudo apt install libcamera0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libcamera0 is already the newest version (0~git20221018+7c855784-1).
libcamera0 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

ldconfig returns nothing:
`pi@BabyPi:~ $ sudo ldconfig`

after reboot, the issue is still the same
`pi@PiBear:~/picam $ ./picam --alsadev hw:1,0
./picam: error while loading shared libraries: libcamera.so.0: cannot open shared object file: No such file or directory`

Do you have any other idea?

iizukanao, 2022-10-22:
@leifnagtegaal Sorry for the inconvenience. The issue has been fixed in [version 2.0.8](https://github.com/iizukanao/picam/releases/tag/v2.0.8). Please use it.

closed 2022-10-22 by iizukanao

leifnagtegaal, 2022-10-23:
Hi Nao,
sorry for bothering you again.
First, i really apreciate your project here and your effort for helping me.

There is a small  issue in the readme.md:

wget https://github.com/iizukanao/picam/releases/download/v2.0.8/_picam-2.0.7-_`uname -m`.tar.xz
i guess, it should look like that
wget https://github.com/iizukanao/picam/releases/download/v2.0.8/picam-2.0.8-`uname -m`.tar.xz

I was not sure, if i simply can download the new build, so i deleted the picam folder and installed it from scratch.
After installing, i get the next issue...
```
pi@PiBear:~/picam $ ./picam --alsadev hw:1,0
./picam: error while loading shared libraries: libepoxy.so.0: cannot open shared object file: No such file or directory
 
```

In the repository, u as not able to find that libepoxy package

iizukanao, 2022-10-24:
Thank you for letting me know about that. I will fix the README.
Does `sudo apt install libepoxy0` resolve the error?

leifnagtegaal, 2022-10-24:
I tried it with just 'libepoxy', and apt returned, that no package is found also with the regex.
But with libepoxy0 i was able to install it.

Next issue while starting picam is:
error: ./state is a broken symbolic link

iizukanao, 2022-10-24:
> error: ./state is a broken symbolic link

Just fix those symbolic links or remove them with `rm -f rec state hooks`

leifnagtegaal, 2022-10-25:
Hi Nao,
thanks for your help.
Now it is working.
Thanks again for sharing this great project with us!

iizukanao, 2022-10-27:
I'm glad to hear that!