iizukanao/picam #188: Trying to build v2.0.8 on Raspi with Debian Buster, getting errors

HWiese1980, 2022-11-09:
I'm trying to build picam v2.0.8 on a Raspberry Pi running Debian Buster. I'm getting the following error messages upon make: 

```bash
pi@raspberrypi:~/picam/build $ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No previous build - default to Release build
-- Platform: armhf
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for module 'libcamera'
--   Found libcamera, version 0.0.0
-- libcamera library found:
--     version: 0.0.0
--     libraries: /usr/lib/arm-linux-gnueabihf/libcamera.so;/usr/lib/arm-linux-gnueabihf/libcamera-base.so
--     include path: /usr/include/libcamera
-- LIBDRM display mode will be unavailable!
-- EGL display mode will be unavailable!
-- Checking for modules 'harfbuzz;fontconfig'
--   Found harfbuzz, version 2.3.1
--   Found fontconfig, version 2.13.1
-- Checking for modules 'libavformat;libavcodec'
--   Found libavformat, version 58.20.100
--   Found libavcodec, version 58.35.100
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/picam/build
pi@raspberrypi:~/picam/build $ make -j
Scanning dependencies of target mpegts
Scanning dependencies of target audio
Scanning dependencies of target muxer
Scanning dependencies of target libstate
Scanning dependencies of target rtsp
Scanning dependencies of target picam_option
Scanning dependencies of target log
Scanning dependencies of target preview
Scanning dependencies of target libhook
[  3%] Building C object libstate/CMakeFiles/libstate.dir/state.c.o
[  6%] Building C object mpegts/CMakeFiles/mpegts.dir/mpegts.c.o
[  9%] Building CXX object audio/CMakeFiles/audio.dir/audio.cpp.o
[ 12%] Building CXX object muxer/CMakeFiles/muxer.dir/muxer.cpp.o
[ 15%] Building CXX object picam_option/CMakeFiles/picam_option.dir/picam_option.cpp.o
[ 18%] Building C object rtsp/CMakeFiles/rtsp.dir/rtsp.c.o
[ 21%] Building C object log/CMakeFiles/log.dir/log.c.o
[ 25%] Building C object libhook/CMakeFiles/libhook.dir/hook.c.o
[ 31%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
[ 31%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
/home/pi/picam/mpegts/mpegts.c: In function ‘setup_audio_stream’:
/home/pi/picam/mpegts/mpegts.c:95:5: error: unknown type name ‘AVChannelLayout’
     AVChannelLayout ch_layout = AV_CHANNEL_LAYOUT_STEREO;
     ^~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [mpegts/CMakeFiles/mpegts.dir/build.make:63: mpegts/CMakeFiles/mpegts.dir/mpegts.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:582: mpegts/CMakeFiles/mpegts.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 34%] Linking C static library libstate.a
[ 37%] Linking C static library librtsp.a
[ 40%] Linking C static library liblog.a
[ 40%] Built target libstate
[ 43%] Linking C static library libhook.a
[ 43%] Built target rtsp
[ 43%] Built target log
[ 43%] Built target libhook
In file included from /home/pi/picam/audio/audio.cpp:8:
/home/pi/picam/audio/audio.hpp:22:28: error: ‘std::function’ has not been declared
   void set_encode_callback(std::function<void(int64_t pts, uint8_t *data, int size, int stream_index, int flags)> callback);
                            ^~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [audio/CMakeFiles/audio.dir/build.make:63: audio/CMakeFiles/audio.dir/audio.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:609: audio/CMakeFiles/audio.dir/all] Error 2
[ 46%] Linking CXX static library libpreview.a
[ 46%] Built target preview
/home/pi/picam/muxer/muxer.cpp: In member function ‘void* Muxer::rec_thread_stop(int)’:
/home/pi/picam/muxer/muxer.cpp:105:11: error: ‘%s’ directive output may be truncated writing up to 1279 bytes into a region of size between 1024 and 2047 [-Werror=format-truncation=]
           "%s/%s", cwd, recording_archive_filepath);
           ^~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
make[2]: *** [muxer/CMakeFiles/muxer.dir/build.make:63: muxer/CMakeFiles/muxer.dir/muxer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:636: muxer/CMakeFiles/muxer.dir/all] Error 2
[ 50%] Linking CXX static library libpicam_option.a
[ 50%] Built target picam_option
make: *** [Makefile:152: all] Error 2
pi@raspberrypi:~/picam/build $
```

Can anyone tell me what's causing these?

Thank you!

iizukanao, 2022-11-10:
Did you install ffmpeg as described in the build instructions? https://github.com/iizukanao/picam/blob/master/BUILDING.md#build-and-install-ffmpeg

If you installed ffmpeg with `apt` or `apt-get`, please uninstall it first.

HWiese1980, 2022-11-10:
Apparently not. I did not even notice the BUILDING.md. Thanks for the link. I'll see if that helps.

HWiese1980, 2022-11-12:
Sad news, I've decided to mothball the project of a self-built surveillance camera for now, and purchased a cheap IP camera instead. All this tinkering to make it work was too much for my wife who just wanted a reliably working camera. So I'll probably not be able to pursue this topic further any time soon - not, unless I'll find another project that needs a dedicated IP camera.

I'm really sorry. I'll close this topic for now... if anyone else runs into the same issue, feel free to open it up again and add all the info that you have. Thanks!

closed 2022-11-12 by HWiese1980