iizukanao/picam #191: Rode USB Mini giving error: cannot set sample format (Invalid argument)

dmonagle, 2022-11-14:
Is there any way to get this microphone working with picam?

Hardware dump looks like this:

```
~/picam $ arecord --dump-hw-params -D hw:1,0
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:1,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S24_3LE
SUBFORMAT:  STD
SAMPLE_BITS: 24
FRAME_BITS: 24
CHANNELS: 1
RATE: 48000
PERIOD_TIME: [1000 1000000]
PERIOD_SIZE: [48 48000]
PERIOD_BYTES: [144 144000]
PERIODS: [2 1024]
BUFFER_TIME: [2000 2000000]
BUFFER_SIZE: [96 96000]
BUFFER_BYTES: [288 288000]
TICK_TIME: ALL
--------------------
arecord: set_params:1343: Sample format non available
Available formats:
- S24_3LE
```

Executing picam with the following: `./picam --rtspout --alsadev hw:1,0` results in:
```
audio setup
sample_rate: 48000
nb_samples: 1024
format: 1
audio_codec_ctx->ch_layout: 4294967297
av_frame->channel_layout: 4294967297
buffer_size=2048
allocated 2048 bytes for audio samples
audio_pts_step: 1920
configuring audio capture device
error: cannot set sample format (Invalid argument)
```

iizukanao, 2022-11-15:
Is it possible to use `plughw:1,0` instead of `hw:1,0`?

dmonagle, 2022-11-15:
That works! Thank you so much.

closed 2022-11-15 by dmonagle

iizukanao, 2022-11-15:
Great!